react-native-camera/react-native-camera #820: Instructions to contribute

diegobfernandez, 2017-08-01:
Hello,

I would love (and need) to contribute to the project, specifically I want to try to add support to Google Mobile Vision API when on Android, but I don't know how to test my changes.

I tried running the Example project but I got no luck. The error it shows is "The development server returned response error code 500

...Unable to resolve module `react` from ..." and it seems to relate to react-native issue 4968.

After following all the suggested steps to fix it still doesn't work and I don't know if I'm doing something wrong.

I strongly suggest to add some guidance on how to get the project up and running.

iegik, 2017-08-04:
Try to downgrade React
Change in Example/package.json to
```
    "react": "~15.4.0",
    "react-native": "0.42.3",
```

skovhus, 2017-10-06:
Related: https://github.com/lwansbrough/react-native-camera/issues/827

sibelius, 2018-01-07:
@diegobfernandez  could u try again with master version?

and also this pr https://github.com/react-native-community/react-native-camera/pull/1071, it has a new example of new implementation of camera module for android

closed 2019-06-08 by stale[bot]