react-native-camera/react-native-camera #830: Camera white when being overlayed

michelalbers, 2017-08-09:
I use react-community/react-navigation for navigating around my app, which is structured like this:

- WrapperNavigator (StackNavigator / Modalmode)
  - AppNavigator (TabNavigator)
    - Camera
    - Gallery
  - LoginAndRegistrationNavigator (StackNavigator)
    - Login
    - Registration
      - Step 1
      - Step 2
      - ....

When the user opens the app and is not logged in, the app will redirect to LoginAndRegistrationNavigator/Login. The Camera is the initial view of AppNavigator and will therefore be rendered and visible until the login overlay appears.

The issue: After the user is done with logging in or registration, the camera will not be working / be white. It will work again, after the user taps the switch camera button once.

Is there a way to tell the camera to reinit? Hiding and showing the view does kinda work but is very bad performancewise.

michelalbers, 2017-08-10:
I tried multiple approaches in force resetting the camera but all failed. Any help?

ArwaLh, 2017-09-07:
I have encountered this same problem. Anyone that could help me?

SimonErm, 2018-09-23:
RCTCamera is deprecated, use RNCamera instead. If this issue still occures, open a new issue please.

closed 2018-09-23 by SimonErm