react-native-camera/react-native-camera #831: Android Barcode Scanning

archieherbias, 2017-08-10:
It seems barcode scanning is not working for me. My code is this one. I am just alerting something if a barcode is detected but I can't get any alerts.

```
class ScannerPage extends Component {
	constructor(props) {
		super(props);
		this.handleBarcodeRead = this.handleBarcodeRead.bind(this);
	}
	handleBarcodeRead(event) {
		Alert.alert('Code detected.');
	}
	render() {
		return (
			<Camera
				{...{
					aspect: Camera.constants.Aspect.fill,
					onBarCodeRead: this.handleBarcodeRead,
				}}
			/>
		);
	}
}
```

Selman555, 2017-08-23:
It should `onBarCodeRead={this.handleBarcodeRead.bind(this)}`
The `.bind(this)` part is redundant as well because you're not accessing component properties (e.g this.state)

archieherbias, 2017-08-24:
But that is the same thing as binding the `handleBarcodeRead` on the constructor and it is the recommended way on how to bind functions.

Binding functions of render has some performance issues because the function is reallocated every re-render.

Selman555, 2017-08-24:
I didn't know about the performance issue. You could of course keep the binding in the constructor but I believe you simple forgot the braces `{}` around your function: `onBarCodeRead: {this.handleBarcodeRead}`

ptomasroos, 2017-09-08:
Is this sorted out according to @Selman555's recommendation @archieherbias ?

archieherbias, 2017-09-09:
No @ptomasroos because it's the same thing with or without the braces. I will just close this as for I am using another library now. Thanks anyway :+1:

closed 2017-09-09 by archieherbias

cinder92, 2017-09-09:
same issue here is not detecting anything in android 6.0

archieherbias, 2017-09-09:
@cinder92 if you want to have Barcode scanning only, this is the library I used
https://www.npmjs.com/package/react-native-smart-barcode

kevinnguy, 2017-09-12:
@archieherbias what's your experience with react-native-smart-barcode? I've been having sporadic performance issues with barcode scanning on Android with this library and I've been looking for a lighter alternative that only focuses on scanning.

archieherbias, 2017-09-12:
@kevinnguy as for my experience, react-native-smart-barcode is pretty fast in terms of scanning for it focuses on scanning only. It uses this library for decoding: https://github.com/zxing/zxing, that's why it seems pretty fast. Though I haven't tested it on ios because I don't have the tools but with Android it seems pretty amazing :+1: