react-native-camera/react-native-camera #832: how make the flash always lighting？？？

yunyouboy, 2017-08-10:


adjustit, 2017-08-17:
Set `flashMode` to `Camera.constants.FlashMode.on` when you initialise the camera's options in your component's `state`.

adjustit, 2017-08-17:
Or if you mean so the light is always illuminated, `Camera.constants.torchMode.on` is what you need.

closed 2018-03-02 by sibelius

christophby, 2018-06-11:
In my case the prop
`flashMode={isTorchOn ? Camera.Constants.FlashMode.torch : Camera.Constants.FlashMode.off}`
did it

n1ru4l, 2018-06-11:
@anniewey which platform are you using?

anniewey, 2018-06-18:
@christophby Tried with your props but still same result.
@n1ru4l I'm using it on Android.

filippoitaliano, 2018-07-06:
Same issue here:

``
import Camera, { RNCamera } from 'react-native-camera';
import React from 'react';
import { StyleSheet, Text, View, Button } from 'react-native';

export default class Photo extends React.Component {
  onBarCodeRead = e => {
    this.props.barcodeCallback(e);
  };

  render() {
    return (
      <View style={styles.container}>
        <Camera
          flashMode={RNCamera.Constants.FlashMode.torch}
          onBarCodeRead={this.onBarCodeRead}
          barCodeTypes={[
            RNCamera.Constants.BarCodeType.qr,
            RNCamera.Constants.BarCodeType.datamatrix,
          ]}
          style={styles.preview}
        />
      </View>
    );
  }
}
``

Flash always off.

xjing97, 2018-12-12:
> In my case the prop
> `flashMode={isTorchOn ? Camera.Constants.FlashMode.torch : Camera.Constants.FlashMode.off}`
> did it
> 
> _flashMode_ is the propname & _Camera.Constants.FlashMode.torch_ is the correct key

Thx, Camera.Constants.FlashMode.torch is the correct key, and it works!!!

> Same issue here:
>
> ```
> import Camera, { RNCamera } from 'react-native-camera';
> import React from 'react';
> import { StyleSheet, Text, View, Button } from 'react-native';
>
> export default class Photo [...]

I think you should change your RNCamera to Camera, and it will work

n1ru4l, 2018-12-12:
`Camera` is deprecated and will be removed in the next major release. Please use `RNCamera`, if you experience any issues with `RNCamera` I encourage you to open a new issue/or fix the issue with a pull request.