react-native-camera/react-native-camera #833: Is it possible to set FPS to a certain value?

XBLDev, 2017-08-11:
Hi 

I'm just wondering if it is possible to set the FPS to a certain value? I want my FPS rate to stay at 30, but it seems that it's never over 30, mostly 29. Is there anyway to set it? 

If i'm not mistaken we are still using hardware.camera? Can we switch to camera2 api? I think with camera2 we can kind of lock FPS rate at a certain value.

willbenmitch, 2017-11-29:
I would also like to be able to take advantage of the hardware.camera2 api. Specifically being able to request highest possible frame rate for the device.

XBLDev, 2017-11-30:
Hi @willbenmitch  Is it still api 1? I haven't used it for a while so I'm not sure.

rsimbu89, 2019-05-16:
Hi @willbenmitch  
Is there an option to set Frames Per Second?

willbenmitch, 2019-05-16:
> Hi @willbenmitch
> Is there an option to set Frames Per Second?

@rsimbu89 I don't know. I'm not a contributor to the project and haven't looked at this in well over a year.

Best of luck.

sibelius, 2019-05-16:
I'm gonna accept a PR that implement this feature

closed 2019-07-22 by stale[bot]