react-native-camera/react-native-camera #834: Add a pause/resume function to the video recording portion.

yoshiquest, 2017-08-11:
I've been using react-native-camera in my mobile app to record video. It has a button that is held to record, released to stop, and held after to start recording again. However, I've had to implement a 1 second minimum recording time to the app, which gets rather annoying. There are 2 reasons for this: I have to merge the videos manually via my own native code, which doesn't seem to properly  recognize videos less than about 1 second long, and that there's currently a bug where attempting to stop the recording to soon (about half a second of recording time or so), will fail to actually stop the recording.

I think that adding a pause/resume call would fix both of these issues. First of all it would remove the need to manually merge the videos, eliminating the first problem. In addition, while a just hunch,  I don't think the aforementioned bug would occur with the pause/resume function.

As such, I'd like to request for a pause/resume call. It should be possible, as AVCaptureFileOutput has a both pauseRecording and resumeRecording instance method (for iOS, not sure about android). If you do implement this, a promise with the current recording length would also be much appreciated.

N1k1tung, 2018-06-07:
pauseRecording/resumeRecording is macOS only: https://developer.apple.com/documentation/avfoundation/avcapturefileoutput/1386806-pauserecording

enisinanaj, 2019-04-13:
@N1k1tung so there's no way to implement this?

closed 2019-06-20 by stale[bot]