react-native-camera/react-native-camera #836: Getting exception ava.lang.RuntimeException: getParameters failed (empty parameters)

angoel, 2017-08-14:
Hi,

I am using 'react-native-camera' and 'react-native-image-crop-picker' in my app. In one screen I using camera to capture and crop the image to feed it to OCR library 'react-native-tesseract-ocr 'and in other screen I am using camera to do barcode reading.

When I am switching from one screen to other. my app is crashing and I am seeing following exception.

Thanks,
Anil

java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$3.done(AsyncTask.java:309)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
	at java.lang.Thread.run(Thread.java:818)
Caused by: java.lang.RuntimeException: getParameters failed (empty parameters)
	at android.hardware.Camera.native_getParameters(Native Method)
	at android.hardware.Camera.getParameters(Camera.java:1996)
	at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:307)
	at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:292)
	at android.os.AsyncTask$2.call(AsyncTask.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
	... 4 more

brunolemos, 2017-08-31:
I have this too, #531 might be a fix

[image]

DeividVeloso, 2017-11-07:
I'm using this verison  "react-native-camera": "^0.10.0", and I have same error yet.

[image]

some fix for this?

RuntimeException: getParameters failed (empty parameters)
    at android.hardware.Camera.native_getParameters(Camera.java)
    at android.hardware.Camera.getParameters(Camera.java:1974)
    at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:307)
    at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:292)
    at android.os.AsyncTask$2.call(AsyncTask.java:295)
...
(5 additional frame(s) were not displayed)

RuntimeException: An error occurred while executing doInBackground()
    at android.os.AsyncTask$3.done(AsyncTask.java:309)
    at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
    at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
    at java.util.concurrent.FutureTask.run(FutureTask.java:242)
    at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)

cesargdm, 2017-11-18:
Same here
RNC: 0.12.0

``` Java
11-18 14:13:55.466 9388-9421/com.aidmedicalid E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #3
                                                                Process: com.aidmedicalid, PID: 9388
                                                                java.lang.RuntimeException: An error occurred while executing doInBackground()
                                                                    at android.os.AsyncTask$3.done(AsyncTask.java:318)
                                                                    at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
                                                                    at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
                                                                    at java.util.concurrent.FutureTask.run(FutureTask.java:242)
                                                                    at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243)
                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
                                                                    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
                                                                    at java.lang.Thread.run(Thread.java:761)
                                                                 Caused by: java.lang.RuntimeException: getParameters failed (empty parameters)
                                                                    at android.hardware.Camera.native_getParameters(Native Method)
                                                                    at android.hardware.Camera.getParameters(Camera.java:3013)
                                                                    at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:312)
                                                                    at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:294)
                                                                    at android.os.AsyncTask$2.call(AsyncTask.java:304)
                                                                    at java.util.concurrent.FutureTask.run(FutureTask.java:237)
                                                                    at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:243) 
                                                                    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133) 
                                                                    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607) 
                                                                    at java.lang.Thread.run(Thread.java:761) 
```

cesargdm, 2017-11-18:
Workaround by setting a default value for width and height in RTCCameraViewFinder

```
            // TODO fix
//            Camera.Size size = camera.getParameters().getPreviewSize();

            int width = 1;
            int height = 1;
```

PichuChen, 2017-12-10:
@cesargdm  but it will make QR Scanner not work.

condorman, 2017-12-14:
Hi, i have same error on: 
Android: 7.0
Model: Samsung - SM-A320FL

`Exception java.lang.RuntimeException: getParameters failed (empty parameters)

android.hardware.Camera.native_getParameters (Camera.java)
android.hardware.Camera.getParameters (Camera.java:1952)
com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent (RCTCameraViewFinder.java:387)
android.view.View.dispatchTouchEvent (View.java:10779)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
android.view.ViewGroup.dispatchTransformedTouchEvent (ViewGroup.java:2864)
android.view.ViewGroup.dispatchTouchEvent (ViewGroup.java:2491)
com.android.internal.policy.DecorView.superDispatchTouchEvent (DecorView.java:605)
com.android.internal.policy.PhoneWindow.superDispatchTouchEvent (PhoneWindow.java:1895)
android.app.Activity.dispatchTouchEvent (Activity.java:3241)
com.android.internal.policy.DecorView.dispatchTouchEvent (DecorView.java:567)
android.view.View.dispatchPointerEvent (View.java:11008)
android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent (ViewRootImpl.java:5155)
android.view.ViewRootImpl$ViewPostImeInputStage.onProcess (ViewRootImpl.java:5007)
android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:4532)
android.view.ViewRootImpl$InputStage.onDeliverToNext (ViewRootImpl.java:4585)
android.view.ViewRootImpl$InputStage.forward (ViewRootImpl.java:4551)
android.view.ViewRootImpl$AsyncInputStage.forward (ViewRootImpl.java:4684)
android.view.ViewRootImpl$InputStage.apply (ViewRootImpl.java:4559)
android.view.ViewRootImpl$AsyncInputStage.apply (ViewRootImpl.java:4741)
android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:4532)
android.view.ViewRootImpl$InputStage.onDeliverToNext (ViewRootImpl.java:4585)
android.view.ViewRootImpl$InputStage.forward (ViewRootImpl.java:4551)
android.view.ViewRootImpl$InputStage.apply (ViewRootImpl.java:4559)
android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:4532)
android.view.ViewRootImpl.deliverInputEvent (ViewRootImpl.java:7092)
android.view.ViewRootImpl.doProcessInputEvents (ViewRootImpl.java:7024)
android.view.ViewRootImpl.enqueueInputEvent (ViewRootImpl.java:6985)
android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent (ViewRootImpl.java:7202)
android.view.InputEventReceiver.dispatchInputEvent (InputEventReceiver.java:185)
android.os.MessageQueue.nativePollOnce (MessageQueue.java)
android.os.MessageQueue.next (MessageQueue.java:323)
android.os.Looper.loop (Looper.java:136)
android.app.ActivityThread.main (ActivityThread.java:6776)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1518)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1408)`

jqn, 2018-06-26:
I have a similar situation on a Samsung S6 when instantiating a the camera in a different screen and rotating

```
06-26 08:23:19.071 22121-22121/com.dealerslink.android.vinscanner E/unknown:React: Exception in native call
    java.lang.RuntimeException: getParameters failed (empty parameters)
        at android.hardware.Camera.native_getParameters(Native Method)
        at android.hardware.Camera.getParameters(Camera.java:1952)
        at com.ekreutz.barcodescanner.camera.CameraSource.setRotation(CameraSource.java:1082)
        at com.ekreutz.barcodescanner.camera.CameraSourcePreview.previewLayout(CameraSourcePreview.java:156)
        at com.ekreutz.barcodescanner.camera.CameraSourcePreview.onLayout(CameraSourcePreview.java:142)
        at android.view.View.layout(View.java:18804)
        at android.view.ViewGroup.layout(ViewGroup.java:5951)
        at com.ekreutz.barcodescanner.ui.BarcodeScannerView.onLayout(BarcodeScannerView.java:265)
        at android.view.View.layout(View.java:18804)
        at android.view.ViewGroup.layout(ViewGroup.java:5951)
        at com.facebook.react.uimanager.NativeViewHierarchyManager.updateLayout(NativeViewHierarchyManager.java:199)
        at com.facebook.react.uimanager.NativeViewHierarchyManager.updateLayout(NativeViewHierarchyManager.java:184)
        at com.facebook.react.uimanager.UIViewOperationQueue$UpdateLayoutOperation.execute(UIViewOperationQueue.java:123)
        at com.facebook.react.uimanager.UIViewOperationQueue$2.run(UIViewOperationQueue.java:782)
        at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:829)
        at com.facebook.react.uimanager.UIViewOperationQueue.access$1500(UIViewOperationQueue.java:44)
        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:868)
        at com.facebook.react.uimanager.GuardedChoreographerFrameCallback.doFrame(GuardedChoreographerFrameCallback.java:32)
        at com.facebook.react.uimanager.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:131)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:925)
        at android.view.Choreographer.doCallbacks(Choreographer.java:702)
        at android.view.Choreographer.doFrame(Choreographer.java:635)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:913)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6682)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
```

JslinSir, 2018-10-19:
getParameters failed (empty parameters)

com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:428)

android.hardware.Camera.native_getParameters(Native Method)
--
2 android.hardware.Camera.getParameters(Camera.java:3243)
3 com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent(RCTCameraViewFinder.java:428)
4 android.view.View.dispatchTouchEvent(View.java:10215)
5 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
6 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
7 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
8 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
9 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
10 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
11 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
12 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
13 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
14 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
15 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
16 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
17 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
18 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
19 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
20 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
21 android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:2765)
22 android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2337)
23 com.android.internal.policy.DecorView.superDispatchTouchEvent(DecorView.java:471)
24 com.android.internal.policy.PhoneWindow.superDispatchTouchEvent(PhoneWindow.java:1864)
25 android.app.Activity.dispatchTouchEvent(Activity.java:3227)
26 com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:425)
27 android.view.View.dispatchPointerEvent(View.java:10446)
28 android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5421)
29 android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5220)
30 android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4677)
31 android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4737)
32 android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4696)
33 android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:4865)
34 android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4704)
35 android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:4922)
36 android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4677)
37 android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:4737)
38 android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:4696)
39 android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:4704)
40 android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4677)
41 android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:7398)
42 android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:7372)
43 android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:7315)
44 android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:7574)
45 android.view.ViewRootImpl$ColorWindowInputEventReceiver.onInputEvent(ViewRootImpl.java:9130)
46 android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
47 android.os.MessageQueue.nativePollOnce(Native Method)
48 android.os.MessageQueue.next(MessageQueue.java:328)
49 android.os.Looper.loop(Looper.java:173)
50 android.app.ActivityThread.main(ActivityThread.java:6802)
51 java.lang.reflect.Method.invoke(Native Method)
52 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1103)
53 com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

Domino987, 2019-01-14:
Any news on this. I occurred at my device, if the device was locked for a certain amount of time.

closed 2019-04-18 by sibelius