react-native-camera/react-native-camera #839: Preview photo and video after capturing

pongponglau, 2017-08-16:
After i take a picture or a video, i want to preview it first before uploading to the server.

May i ask if it is possible to do this with react-native-camera? If so, can you show me how to do it?

Thank you very much

adjustit, 2017-08-17:
The easiest way to implement it is to use a callback inside of the `capture` method which updates your component's state with the image's `path` and a flag to say that you're previewing that toggles between the camera view and an image view (using conditional block rendering in your component which triggers based on a boolean state).

The code is fairly simple once you understand the logic.

gitlovenotwar, 2017-09-01:
@adjustit How are you going to do that with video? showing a thumbnail and play it back?

adjustit, 2017-09-03:
@gitlovenotwar we removed video from our app, but previously I believe that was accomplished with a custom Video component.

danilags, 2017-09-20:
Just create a new component to preview the image, and make sure the data from the promise after taking the picture is pass to the new component. The way it works with me.. 💯

rohitgoyal, 2018-01-10:
This is how I am trying to preview image 

`{this.state.docFrontPicUri && <Image source={{uri:this.state.docFrontPicUri, isStatic:true}} style={{width:global.constants.APP_WIDTH,height:global.constants.APP_HEIGHT*0.7}} />}`

And this is my takePicture function.

```
takePicture() {
    const options = {};
    this.camera.capture({metadata: options})
      .then((data) => this.setState({
        docFrontPicUri:data.path
      }))
      .catch(err => console.error(err));
  }
```

but when i take a snap i get the error 

> ExceptionsManager.js:73 No suitable image URL loader found for assets-library://asset/asset.JPG?id=6F33CD35-2B07-4BA2-BF1B-52B7CA210F28&ext=JPG

Can someone help how they have shown the preview of the snap after capturing?

kanteankit, 2018-03-03:
@adjustit I am trying your approach of conditional block rendering. The issue I face is that I see a white screen for a fraction of second, when the state changes from from camera to preview. Has anyone else faced this issue? Am I doing something wrong?

chmielot, 2018-03-03:
@kanteankit Can you share the code for further investigation?

sibelius, 2018-03-03:
FIxed on RNCamera

closed 2018-03-03 by sibelius

chmielot, 2018-03-03:
I am able to reproduce this issue with RNCamera, however, I'm not sure if this is RNCamera related. The screen I see is actually the backgroundColor of the container. Like react-native renders the container and it takes some time to display the image. 

```
// Snip
import { RNCamera as Camera } from 'react-native-camera';

const styles = StyleSheet.create({
  container: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
    backgroundColor: '#000',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
    height: Dimensions.get('window').height,
    width: Dimensions.get('window').width
  },
  capture: {
    width: 70,
    height: 70,
    borderRadius: 35,
    borderWidth: 5,
    borderColor: '#FFF',
    marginBottom: 15,
  },
  cancel: {
    position: 'absolute',
    right: 20,
    top: 20,
    backgroundColor: 'transparent',
    color: '#FFF',
    fontWeight: '600',
    fontSize: 17,
  }
});

class TakePictureScreen extends Component {

  constructor(props) {
    super(props);
    this.state = {
      path: null,
    };
  }

  takePicture = async () => {
    try {
      const data = await this.camera.takePictureAsync();
      this.setState({ path: data.uri });
      // this.props.updateImage(data.uri);
      // console.log('Path to image: ' + data.uri);
    } catch (err) {
      console.log('err: ', err);
    }
  };

  renderCamera() {
    return (
      <Camera
        ref={(cam) => {
          this.camera = cam;
        }}
        style={styles.preview}
        flashMode={Camera.Constants.FlashMode.off}
        permissionDialogTitle={'Permission to use camera'}
        permissionDialogMessage={'We need your permission to use your camera phone'}
      >
        <TouchableHighlight
          style={styles.capture}
          onPress={this.takePicture.bind(this)}
          underlayColor="rgba(255, 255, 255, 0.5)"
        >
          <View />
        </TouchableHighlight>
      </Camera>
    );
  }

  renderImage() {
    return (
      <View>
        <Image
          source={{ uri: this.state.path }}
          style={styles.preview}
        />
        <Text
          style={styles.cancel}
          onPress={() => this.setState({ path: null })}
        >Cancel
        </Text>
      </View>
    );
  }

  render() {
    return (
      <View style={styles.container}>
        {this.state.path ? this.renderImage() : this.renderCamera()}
      </View>
    );
  }
}
```

omitobi, 2018-03-13:
@chmielot Your snippet helped implement the preview for a taken picture. As for me, I saw a black background before the image got rendered. I use RCTCamera.

mdelclaro, 2019-03-28:
@chmielot I used your code as well, but the image preview is changing the orientation. Any ideia why?

sibelius, 2019-03-28:
this is related to exif

leleofg, 2019-04-23:
This can also help, using the new solution with hooks and permission to android with save to gallery with CameraRoll.

`export default Camera = ({ navigation }) => {

  const [imageUri, setImageUri] = useState(null);
  const [type, setTypeCamera] = useState(RNCamera.Constants.Type.back);

  submitPicture = async () => {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE,
        {
          "title": "Access Storage",
          "message": "Access Storage for the pictures"
        }
      )
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        await CameraRoll.saveToCameraRoll(imageUri);
      } else {
        console.log("Permissao de camera negada.");
      }
    } catch (err) {
      console.warn(err);
    }

    alert("Image save with success");
  }

  takePicture = async () => {
    try {
      if (this.camera) {
        const options = {
          quality: 0.5,
          base64: true,
          forceUpOrientation: true,
          fixOrientation: true
        };

        const { uri } = await this.camera.takePictureAsync(options);
        setImageUri(uri);
      }
    } catch (err) {
      alert(err.message);
    }
  };

  changeTypeCamera = () => {

    if (type === RNCamera.Constants.Type.back) {
      setTypeCamera(RNCamera.Constants.Type.front);
    } else {
      setTypeCamera(RNCamera.Constants.Type.back);
    }
  }

  return (
    imageUri ?
      <ImageBackground
        style={{ width: "100%", height: "100%" }}
        source={{ uri: imageUri }}>
        <MagicButton title="Enviar para" bg="#fff" colorText="#000" method={() => submitPicture()} />
        <TouchableOpacity onPress={() => setImageUri(null)}>
          <Text style={{ textAlign: "center" }}>
            X
          </Text>
        </TouchableOpacity>
      </ImageBackground>
      :
      <RNCamera
        ref={camera => {
          this.camera = camera;
        }}
        style={styles.camera}
        type={type}
        autoFocus={RNCamera.Constants.AutoFocus.on}
        flashMode={RNCamera.Constants.FlashMode.off}
        permissionDialogTitle={"Permission to use camera"}
        permissionDialogMessage={
          "We need your permission to use your camera phone"
        }>
        <SafeAreaView>
          <View style={styles.header}>
          </View>
        </SafeAreaView>
        <SafeAreaView>
          <View style={styles.footer}>
            <Icon name="times" size={25} color="#fff" onPress={() => navigation.goBack()} />
            <TouchableOpacity onPress={takePicture} >
              <View style={styles.snapButton}>
                <View style={styles.innerSnapButton}>
                  <Icon name="camera" size={25} color="white" />
                </View>
              </View>
            </TouchableOpacity>
            <Icon name="undo" size={25} color="#fff" onPress={() => changeTypeCamera()} />
          </View>
        </SafeAreaView>
      </RNCamera>
  )
}`

leleofg, 2019-04-23:
This can also help, using the new solution with hooks and permission android to save in gallery with CameraRoll.

```
export default Camera = ({ navigation }) => {

  const [imageUri, setImageUri] = useState(null);
  const [type, setTypeCamera] = useState(RNCamera.Constants.Type.back);

  submitPicture = async () => {
    try {
      const granted = await PermissionsAndroid.request(
        PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE,
        {
          "title": "Access Storage",
          "message": "Access Storage for the pictures"
        }
      )
      if (granted === PermissionsAndroid.RESULTS.GRANTED) {
        await CameraRoll.saveToCameraRoll(imageUri);
      } else {
        console.log("Permissao de camera negada.");
      }
    } catch (err) {
      console.warn(err);
    }

    alert("Image save with success");
  }

  takePicture = async () => {
    try {
      if (this.camera) {
        const options = {
          quality: 0.5,
          base64: true,
          forceUpOrientation: true,
          fixOrientation: true
        };

        const { uri } = await this.camera.takePictureAsync(options);
        setImageUri(uri);
      }
    } catch (err) {
      alert(err.message);
    }
  };

  changeTypeCamera = () => {

    if (type === RNCamera.Constants.Type.back) {
      setTypeCamera(RNCamera.Constants.Type.front);
    } else {
      setTypeCamera(RNCamera.Constants.Type.back);
    }
  }

  return (
    imageUri ?
      <ImageBackground
        style={{ width: "100%", height: "100%" }}
        source={{ uri: imageUri }}>
        <MagicButton title="Enviar para" bg="#fff" colorText="#000" method={() => submitPicture()} />
        <TouchableOpacity onPress={() => setImageUri(null)}>
          <Text style={{ textAlign: "center" }}>
            X
          </Text>
        </TouchableOpacity>
      </ImageBackground>
      :
      <RNCamera
        ref={camera => {
          this.camera = camera;
        }}
        style={styles.camera}
        type={type}
        autoFocus={RNCamera.Constants.AutoFocus.on}
        flashMode={RNCamera.Constants.FlashMode.off}
        permissionDialogTitle={"Permission to use camera"}
        permissionDialogMessage={
          "We need your permission to use your camera phone"
        }>
        <SafeAreaView>
          <View style={styles.header}>
          </View>
        </SafeAreaView>
        <SafeAreaView>
          <View style={styles.footer}>
            <Icon name="times" size={25} color="#fff" onPress={() => navigation.goBack()} />
            <TouchableOpacity onPress={takePicture} >
              <View style={styles.snapButton}>
                <View style={styles.innerSnapButton}>
                  <Icon name="camera" size={25} color="white" />
                </View>
              </View>
            </TouchableOpacity>
            <Icon name="undo" size={25} color="#fff" onPress={() => changeTypeCamera()} />
          </View>
        </SafeAreaView>
      </RNCamera>
  )
}
```