react-native-camera/react-native-camera #843: Understanding captureTarget properties

longnguyennn, 2017-08-18:
Hi,
Can someone explain to me what does `Camera.constants.CaptureTarget.disk` and `Camera.constants.CaptureTarget.temp` do? I understand that `CaptureTarget.disk` will save the photo to the disk (phone's disk?), but then how can I find it on my phone? 
Also, when I choose the option `Camera.constants.CaptureTarget.cameraRoll`, it saves the photo in DCIM folder. Then what should I do f I want to save the photo to a different folder? 
Thanks in advance!!

milon87, 2017-08-18:
Implementing in Android:
 captureTarget={Camera.constants.CaptureTarget.disk}
  saves the picture to "file:///storage/emulated/0/Pictures/file_name.jpg"

captureTarget={Camera.constants.CaptureTarget. cameraRoll}
  saves the picture to "file:///storage/emulated/0/DCIM/file_name.jpg"

captureTarget={Camera.constants.CaptureTarget. temp}
  saves the picture to "file:///data/data/{package_name}/cache/file_name.jpg"

longnguyennn, 2017-08-28:
I've read the issue but they didn't mention how to save captured photos to a different folder. What should I do if I want to save captured images to a folder named `XYZ` instead of `DCIM` (on Android) ?

adjustit, 2017-08-29:
@longnguyennn you will probably need to use another library's methods on top of this one, like [react-native-fs](https://github.com/itinance/react-native-fs)'s `moveFile`.

ptomasroos, 2017-09-08:
Thanks for helping out @adjustit

closed 2017-09-08 by ptomasroos