react-native-camera/react-native-camera #852: Get actual path of image?

mchandleraz, 2017-08-25:
First off, thanks for this component!

Is it possible to get the path of the image rather than the assets-library path? I'm trying to upload the captured image to a 3rd party API. Using another react-native-image-crop-picker, I can get a uri that looks something like `/Users/****/Library/Developer/CoreSimulator/Devices/8BDCD3C7-F127-4162-90B8-5E044029495C/data/Containers/Data/Application/324B1254-4638-46F6-B695-92A6BB3FE6DA/tmp/react-native-image-crop-picker/0A31CED0-995F-4CDF-9A9A-1DB6216EB621.jpg` (this is of course while running on simulator). Using this URI, I can upload to my 3rd party API. When I try to upload using the assets-library path returned from react-native-camera, the SDK for the 3rd party API throws an error.

Is it possible to get a URI that looks like the one provided from react-native-image-crop-picker?

adjustit, 2017-08-29:
How are you actually submitting it to the API?

If you're using `fetch`, are you loading the file into a multipart `FormData` object? We use this extensively within our app with perfect success. Our code (summarised, obviously) looks like:

```
const body = new FormData();
const submitUrl = apiUrl;

let file = {
	uri: image.path,
	name: 'name.jpg',
	type: 'image/jpg'
};

body.append('fname', file);
			
fetch(submitUrl, {
	method: 'POST',
	body: body
}).then((response) => response.json())
.then(function(responseData) {
	...
}).catch(function(err) {
	...
});
```

It's also important that within `name` in this usage, you must supply the file extension (in this case, `.jpg`) or it will be rejected.

mchandleraz, 2017-08-29:
Thanks for the input @adjustit. Unfortunately, the pattern you're using won't work for my use case as I'm using the Cloudinary SDK. I may migrate to using their REST API for this feature though, as I'd be able to use your suggestion if I'm not using their SDK.

adjustit, 2017-08-30:
If you can work out how to use relative pathing, the filename should be a combination of the `id` and `ext` parameters that are provided by the `path` response (ie. `assets-library://asset/asset.JPG?id=575ABD7C-3DDD-47E6-AD12-F4918D7B211F&ext=JPG` would be stored as `575ABD7C-3DDD-47E6-AD12-F4918D7B211F.jpg` in the local folder).

SimonErm, 2018-09-23:
RCTCamera is deprecated, use RNCamera instead.

closed 2018-09-23 by SimonErm