react-native-camera/react-native-camera #853: onBarCodeRead make app exit.

DeFenceD, 2017-08-25:
when i use react-navigation recall the scan router.
if i set onBarCodeRead , when i do the first  time to scan it is ok, and i make it leave the view.
but, when i second into scan view, it will make me exit my app.
i check the cause, i found when i just set the props 'onBarCodeRead' can encounter this problem.

this is my camera and my navigation.
`<Camera ref='scan' style={css.preview}  onBarCodeRead={this.methods.scan}  aspect={Camera.constants.Aspect.fill}>`
`navigation.navigate('Scan', {task: 'scanAuthCode'});`

closed 2017-09-04 by DeFenceD

kerwin1, 2017-10-26:
I have the same problem,Can anyone help me?

DeFenceD, 2017-10-29:
I see you look like Chinese, so I write Chinese first, if you can't read then I support English.
关键时要手动从路由记录数组中将Scan这个视图从该数组删除。
你可以监听去路由变化，或者在你想要需要删除的时候。
这样做完之后，就可以debug闪退的问题。