react-native-camera/react-native-camera #854: SIGABRT on ios when barCodeTypes specified

deedoc, 2017-08-25:
Hi guys!

I'm using "react-native-camera": "0.10.0" and specified barCodeTypes like this
`barCodeTypes={["qr"]}`

It works nice on android, but i began to get SIGABRT on ios.

In XCode debug i see that when i leave barCodeTypes blank it passed into
`[metadataOutput setMetadataObjectTypes:self.barCodeTypes];`
as array of AVFoundation qualifier strings ["org.iso.QRCode", ...]

But if i do specify barCodeTypes it passes as is - ["qr", ...] - this is reason of SIGABRT

I avoided this SIGABRT with following code
```
const BAR_CODE_TYPES = [
  Platform.select({android: "qr", ios: "org.iso.QRCode"})
];
```

srolija, 2017-08-26:
Had the same issue, you should use constant rather than string, try like this
`barCodeTypes={[Camera.constants.BarCodeType.qr]}`.

norberdo, 2017-11-10:
Correct, this fixed my problem, thank you!

closed 2019-06-08 by stale[bot]