react-native-camera/react-native-camera #855: onBarCodeRead when read Code128 in android 4.4,it doesn‘ work

wenwen1995, 2017-08-25:
When I use the camera to scan the code128,it works good in android 5+,but when I use it in android ,it only can scan qr,but don't read the code128 data,what's wrong ??

wenwen1995, 2017-08-28:
Okay,I find the problem,it's not the code problem,it may be the code128.....When I use android 4.4 phone to scan the pc code128 photos,like this,
[image]

and I scan many of these photos,it works well,and the datas also appear....I think it may be not  sensitive..

rajeshkos, 2017-08-31:
Hi @wenwen1995 

Can you please send me sample code for barcode reader for android and ios ?

iegik, 2017-09-05:
It is ean13

closed 2019-08-07 by stale[bot]