react-native-camera/react-native-camera #860: How can I increase the speed of the render to react native camera?

MackayBravo, 2017-08-27:
I use the react navigation module for routing, when I go to screen of module react native camera fps drops to 2. How to solve this problem?
THANKS
IT IS VERY IMPORTANT !!

adjustit, 2017-08-29:
The only real ways to somewhat solve it that I know of are to either, a) preload the component you're switching to and then show the camera with a `state` toggle, or b) get faster hardware. 

Initialising the camera (whether it's the native camera built into the OS or through an abstracted layer like RN with this library) is inherently expensive and will consume a lot of your resources.

Also, sidenote: ALL-CAPS and exclamation marks make it much less likely for people to want to help you. Especially when you already have an existing issue for the problem mentioned on the repo you're asking on.

p-ugulino, 2018-01-10:
@adjustit hey, could you elaborate more on how to preload a component?

n1ru4l, 2018-12-19:
Please open a new issue if you are still experiencing these issues with the newest version of `RNCamera`.

closed 2018-12-19 by n1ru4l