react-native-camera/react-native-camera #865: Consistency Isuues in barcode reading

Megamind2301, 2017-08-30:
I am using react-native-barcodescanner(react-native-camera dependent) in one of my projects.

The project demands app to be open all the time and should scan every incoming profile's barcode, read that and show popup of that profile.
While the app is open response is instant for quite some time but when app is ideal for some time, it stops reading barcode. If I close the app and reopen it, everything starts working smoothly again.

I wanted to know what is causing that and how to solve it. Or is it library issue!

sharnik, 2017-11-30:
@Megamind2301 Probably a nice place to start debugging the issue would be to drop `react-native-barcodescanner` first (it's not maintained anymore anyway) and see if the issue still happens when using `react-native-camera` directly.

localhosted, 2018-01-24:
I have the same problem, but I'm only using react-native-camera. I can scan QR code but not ean13.

closed 2019-06-08 by stale[bot]