react-native-camera/react-native-camera #872: How to identify local 2D code pictures, such as albums？

yellowJhon, 2017-09-04:
The project can identify pictures captured by the camera, but I did not find the interface to identify the local image. 
How to solve it？ Or this feature is not supported yet?

ptomasroos, 2017-09-08:
Give it a try with react-native-image-picker

closed 2017-09-08 by ptomasroos

yellowJhon, 2017-09-28:
Finally, I decide to link APP to the native platform by developing iOS and Android.

TrustTheBoy, 2017-12-28:
@yellowJhon Hello, how do you implement this function

yellowJhon, 2017-12-28:
Yeah
I'm sorry that I can't paste my source code here for some reasons.
You can refer this link 
[Java Examples for com.google.zxing.RGBLuminanceSource
](https://www.javatips.net/api/com.google.zxing.rgbluminancesource)to get more details.

TrustTheBoy, 2017-12-28:
@yellowJhon  Thank you very much, but it only works on Android. iOS?

yellowJhon, 2017-12-28:
I am not familiar with the iOS development...

TrustTheBoy, 2017-12-28:
Ok, thanks again for your help, I'm looking for it @yellowJhon