react-native-camera/react-native-camera #875: App Crash on Android 7.x

ntnamag, 2017-09-05:
I have a issue, work fine with Android 5.x.x, app force close with Android 7.x

ptomasroos, 2017-09-08:
Have you check the debug log? Pretty sure this has todo with runtime permissions not being requested. Give it a try and let us know if it fixes your issue or not

ntnamag, 2017-09-12:
I fixed it, react-native-camera can't auto request permission android, i have to manual for it

OzoTek, 2017-09-25:
Hey @ntnamag, what do you need to manually do? It used to work on 7.x (can't test on lower versions) and now it just crashes without any log :/

ntnamag, 2017-09-28:
@OzoTek, you have to use [](https://facebook.github.io/react-native/docs/permissionsandroid.html).
with capture a Picture, you can see my code:
`if (Platform.OS === 'android') {
                            try {
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.CAMERA))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA, {
                                        'title': 'Xin quyền truy cập Camera',
                                        'message': 'Túi Xách Eva cần quyền truy cập Camera để bạn có thể chụp ảnh trong ứng dụng'
                                    });
                                }
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE, {
                                        'title': 'Xin quyền truy cập micro',
                                        'message': 'Túi Xách Eva cần quyền truy cập để lưu trữ hình ảnh'
                                    });
                                }
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE, {
                                        'title': 'Xin quyền truy cập micro',
                                        'message': 'Túi Xách Eva cần quyền truy cập để lưu trữ hình ảnh'
                                    });
                                }
                            } catch (err) {

                            }
                        }
                        if ((await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE)) && (await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE)) && (await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.CAMERA))) {
                            this.setState({
                                visibleCamera: true
                            });
                        }`

and with Video Record:
`if (Platform.OS === 'android') {
                            try {
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.CAMERA))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.CAMERA, {
                                        'title': 'Xin quyền truy cập Camera',
                                        'message': 'Túi Xách Eva cần quyền truy cập Camera để bạn có thể chụp ảnh trong ứng dụng'
                                    });
                                }
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.RECORD_AUDIO))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.RECORD_AUDIO, {
                                        'title': 'Xin quyền truy cập micro',
                                        'message': 'Túi Xách Eva cần quyền truy cập mirco để ghi âm khi quay video trong ứng dụng'
                                    });
                                }
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE, {
                                        'title': 'Xin quyền truy cập micro',
                                        'message': 'Túi Xách Eva cần quyền truy cập để lưu trữ hình ảnh'
                                    });
                                }
                                if (!(await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE))) {
                                    let granted = await PermissionsAndroid.request(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE, {
                                        'title': 'Xin quyền truy cập micro',
                                        'message': 'Túi Xách Eva cần quyền truy cập để lưu trữ hình ảnh'
                                    });
                                }
                            } catch (err) {

                            }
                        }
                        if ((await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.WRITE_EXTERNAL_STORAGE)) && (await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.READ_EXTERNAL_STORAGE)) && (await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.CAMERA)) && (await PermissionsAndroid.check(PermissionsAndroid.PERMISSIONS.RECORD_AUDIO))) {
                            this.setState({
                                visibleCameraVideo: true
                            });
                        }`

ntnamag, 2017-09-28:
@OzoTek I'm bad English, you see my code fix it

OzoTek, 2017-09-29:
Thanks @ntnamag, I had this figured out. I actually used https://github.com/yonahforst/react-native-permissions so I can ask permission for both iOS and Android.

By the way you don't have to `check` before doing `request`, because `request` will automatically return `granted` if it's already authorized, so you can remove all these `if` ;)

closed 2018-09-23 by SimonErm