react-native-camera/react-native-camera #878: Camera is being used after camera.release() was called

michelalbers, 2017-09-08:
On Android the Camera occasionally crashes with the following exception:

### Metadata
React Native 0.47.1
React Native Camera 0.10.0

### Strack Trace
```Java
Exception java.lang.RuntimeException: Camera is being used after Camera.release() was called
android.hardware.Camera.setHasPreviewCallback (Camera.java)
android.hardware.Camera.access$600 (Camera.java:152)
android.hardware.Camera$EventHandler.handleMessage (Camera.java:1181)
android.os.Handler.dispatchMessage (Handler.java:102)
android.os.Looper.loop (Looper.java:158)
android.app.ActivityThread.main (ActivityThread.java:7224)
java.lang.reflect.Method.invoke (Method.java)
com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1230)
com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1120)
```

ptomasroos, 2017-09-08:
Can you please provide a reproduction repository?

gitlovenotwar, 2017-09-13:
I have the same problem. When recording a video and stops it. It crashes the app with the same logs.

yourcontainer, 2017-09-26:
@ptomasroos also for me, when you put app to the background

jslok, 2017-10-10:
I'm getting the same error from many users in bugsnag, though I am not able to reproduce it on my own.

mannol, 2017-10-11:
Backtrace from crashylitics:
```Fatal Exception: java.lang.RuntimeException: Camera is being used after Camera.release() was called
       at android.hardware.Camera.lock(Camera.java)
       at com.lwansbrough.RCTCamera.RCTCameraModule.releaseMediaRecorder(RCTCameraModule.java:405)
       at com.lwansbrough.RCTCamera.RCTCameraModule.onError(RCTCameraModule.java:130)
       at android.media.MediaRecorder$EventHandler.handleMessage(MediaRecorder.java:1495)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6682)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1520)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1410)
```

Seems to me that onError is called right after camera is released for some reason. Maybe add a check for that case?

jslok, 2017-10-11:
I found it happens for me when I open the camera then navigate to another screen with the camera. Initializing the camera on two different screens causes a crash every time. Using rn 0.47.2

DeividVeloso, 2017-11-07:
Same error here:

RuntimeException: Camera is being used after Camera.release() was called
    at android.hardware.Camera.setHasPreviewCallback(Camera.java)
    at android.hardware.Camera.-wrap0(Camera.java)
    at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1127)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:148)

Some one  have a fix for this?

Thanks

jslok, 2017-11-07:
I solved the issue by removing the camera component before i navigate to a new screen.

DeividVeloso, 2017-11-07:
@jslok how did you do that?

Can you post your code, please?

jslok, 2017-11-07:
There may be a simpler way of doing this, but this is how I did it with redux.

mapStateToProps passes the screen key from redux, and the constructor sets it into the screen state.
```
const mapStateToProps = ({ nav }) => {
	let screenKey;
	try {
		let currentRouteIndex = nav.routes[nav.index].index;
	    let currentRoute = nav.routes[nav.index].routes[currentRouteIndex].routes[0].index;
		let currentRouteSearch = nav.routes[nav.index].routes[currentRouteIndex].routes[0].routes[currentRoute];
		screenKey = currentRouteSearch.key
	}
	catch(err) {
	    console.log(err);
	}

    return {
		screenKey: screenKey
	}
}
```

```
  constructor (props) {
    super(props)
    this.state = {
	screenKey: props.screenKey
    }
  }
```

Then only show the camera if the current screen key from redux matches the camera's screen key.
```
  renderCamera = () => {
      if (this.state.screenKey !== this.props.screenKey) return null;
      return (
          <View style={styles.camContainer}>
              <Camera
                style={styles.camPreview}
              >
              </Camera>
            </View>
      )
  }
```

And also utitilize shouldComponentUpdate to minimize
```
  shouldComponentUpdate(nextProps, nextState) {
	  if (this.state.screenKey !== this.props.screenKey && this.state.screenKey === nextProps.screenKey) return true; //add camera back if user hits backbutton
	  if (this.state.screenKey === this.props.screenKey && this.state.screenKey !== nextProps.screenKey) return true; //remove camera when navigating to different screen
	  if (this.state !== nextState) return true;
	  return false;
  }
```

jslok, 2017-11-07:
If you do not use redux, you can use simple `this.setState({hideCamera:true})` before you navigate to a new screen.
```
  renderCamera = () => {
      if (this.state.hideCamera) return null;
      return (
          <View style={styles.camContainer}>
              <Camera
                style={styles.camPreview}
              >
              </Camera>
            </View>
      )
  }
```
If the user hits the back button to go back to the camera's screen, the camera will not return however. In that case you can add a button to call `this.setState({hideCamera:false})`

DeividVeloso, 2017-11-08:
@jslok Thanks for your helping, but I made some different

I take the ref of may component and in constructor I set component Camera to null

  constructor(props) {
    super(props);

    this.camera = null;
}

render(){
return(
  <Camera
          ref={cam => {
            this.camera = cam;
          }}
)
}

And in my componentWillUnMount I set to null again:

 componentWillUnmount() {
    this.camera = null;
  }

But my problem is in stacknavigator, because I use a Flow of payment with 3 screen and the first is a Camera.

But in second screen maybe I want return for camera, and I need their in memory in stack, but I want to torn off temporary to economize battery.

jslok, 2017-11-08:
@DeividVeloso Does it solve the issue for you? I tried that before but it didn't work, I think because react-navigation does not unmount the screen when navigating to another screen and the camera stays mounted. Then navigating to a second camera screen caused my app to crash due to having two cameras mounted in the app.

UberMC, 2017-12-04:
@jslok 
> const mapStateToProps = ({ nav }) => {

Where does nav come from? Can you explain how to implement "nav" it's just a null object...

blaues0cke, 2018-01-04:
I have a qr code scanner as app-tab and in a modal overlay. After I opened the modal camera, the one in the tab stopped working. I got a working solution by just replacing the whole qr code tab by a new instance of itself when the `MODAL_CLOSE` event (using redux) occurred: 

```
import * as   ActionTypes         from '../actions/types';
import      { NavigationActions } from 'react-navigation';
import        Registry            from '../lib/Registry';

const TabQrCodeStackNavigationReducer = (state, action) => {
    const TabQrCodeStackNavigator = Registry.get('TabQrCodeStackNavigator');
    const newState                = TabQrCodeStackNavigator.router.getStateForAction(copyiedAction, state);

    let newStateBuffer = newState || state;

    if (action.type === ActionTypes.MODAL_CLOSE) {
        newStateBuffer = {
            index:  0,
            routes: [
                {
                    key:       'TabQrCode' + Math.random(),
                    routeName: 'TabQrCode_QRCodeScanner'
                }
            ]
        };
    }

    return newStateBuffer;
};

module.exports = TabQrCodeStackNavigationReducer;
```

Rewieer, 2018-01-24:
This is actually an issue because when moving to another view with an embedded camera, the old one is not released.
I believe there should be a mecanism to manually release/reopen the camera when moving using the navigator. 

@lwansbrough any thoughts on this ?

sibelius, 2018-01-24:
The problem is that camera is a singleton, we gonna fiz this on RNCamera soon

Rewieer, 2018-01-24:
Do you need any help on this ?

sibelius, 2018-01-24:
you can follow this commit https://github.com/expo/expo/commit/58e732bdbf7e091ab766000b10ff1991a68fce78

ZionChang, 2018-03-02:
Any progress?

sibelius, 2018-03-02:
Use RNCamera

closed 2018-03-02 by sibelius

ZionChang, 2018-03-16:
@sibelius Sometimes also crash when I use RNCamera

sibelius, 2018-03-16:
Open another issue with stacktrace of RNCamera please