react-native-camera/react-native-camera #884: java.lang.OutOfMemoryError

aurelienmrt, 2017-09-11:
Hello, I get this error using your photo module in my Android app with a Samsung Galaxy J5 (2016) on Android 6.0.
Can you give me some ideas to solve this error?

Thank you for your help.

```
java.lang.OutOfMemoryError: 
  at dalvik.system.VMRuntime.newNonMovableArray (Native Method)
  at android.graphics.BitmapFactory.nativeDecodeStream (Native Method)
  at android.graphics.BitmapFactory.decodeStreamInternal (BitmapFactory.java:882)
  at android.graphics.BitmapFactory.decodeStream (BitmapFactory.java:858)
  at android.graphics.BitmapFactory.decodeStream (BitmapFactory.java:896)
  **at com.lwansbrough.RCTCamera.MutableImage.toBitmap (MutableImage.java:133)
  at com.lwansbrough.RCTCamera.MutableImage.<init> (MutableImage.java:36)
  at com.lwansbrough.RCTCamera.RCTCameraModule$3$1.run (RCTCameraModule.java:539)**
  at android.os.AsyncTask$SerialExecutor$1.run (AsyncTask.java:234)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1113)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:588)
  at java.lang.Thread.run (Thread.java:818)
```

fbacker, 2017-09-29:
Does it work if you lower the quality? see captureQuality

aurelienmrt, 2017-10-02:
@fbacker 
Yes, I have the same problem with lower quality.

tobob, 2017-12-04:
https://stackoverflow.com/questions/32244851/androidjava-lang-outofmemoryerror-failed-to-allocate-a-23970828-byte-allocatio/32245018

jgcmarins, 2018-05-21:
@eabait did you try @tobob stackoverflow link solution?

aurelienmrt, 2018-05-22:
Hello @eabait, the problem still appears on some Samsung devices as well.

eabait, 2018-05-28:
Thanks, yes, we are struggling to get it work on several Samsung devices as well. Not easy to replicate since we get reports from users, and we havent able to replicate in our environments

n1ru4l, 2018-12-19:
Please open a new issue if you are still experiencing this problem with the newest version of `RNCamera`.

closed 2018-12-19 by n1ru4l