react-native-camera/react-native-camera #885: Execution failed for task ':app:transformClassesWithDexForDebug'

Naveen-ICARUS, 2017-09-12:
I try to use the react-native-camera by following both automatic and manual installation. 
but always shows this error please help me in .
[image]

jozekcore, 2017-09-14:
i have the same issue

ashishrokr, 2017-09-20:
same issue while installing app

irsyadmhdilham, 2017-11-18:
@ashishrokr I did tweak multiDexEnabled to true and add in the dependencies. Still doesn't solve the issue

n1ru4l, 2018-09-25:
This seems to be solved.

@irsyadmhdilham If you still experience this issue after using the newest npm release and/or the master branch and RNCamera I encourage you to open a new issue.

closed 2018-09-25 by n1ru4l

kapil708, 2018-11-23:
hi @irsyadmhdilham did you find any solution.