react-native-camera/react-native-camera #887: Android recording video error

gitlovenotwar, 2017-09-13:
<pre>
Media recorder stop error.
java.lang.RuntimeException: stop failed.
    at android.media.MediaRecorder._stop(Native Method)
    at android.media.MediaRecorder.stop(MediaRecorder.java:973)
    at com.lwansbrough.RCTCamera.RCTCameraModule.releaseMediaRecorder(RCTCameraModule.java:386)
    at com.lwansbrough.RCTCamera.RCTCameraModule.stopCapture(RCTCameraModule.java:654)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:363)
    at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
    at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
    at android.os.Handler.handleCallback(Handler.java:739)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
    at android.os.Looper.loop(Looper.java:148)
    at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
    at java.lang.Thread.run(Thread.java:818)
09-13 11:17:21.772 11477-16183/com.rywards E/unknown:ReactNative: Exception in native call
    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.media.MediaRecorder.reset()' on a null object reference
        at com.lwansbrough.RCTCamera.RCTCameraModule.releaseMediaRecorder(RCTCameraModule.java:392)
        at com.lwansbrough.RCTCamera.RCTCameraModule.stopCapture(RCTCameraModule.java:654)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:363)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
        at android.os.Looper.loop(Looper.java:148)
        at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
        at java.lang.Thread.run(Thread.java:818)
09-13 11:17:21.772 11477-11477/com.rywards E/AndroidRuntime: FATAL EXCEPTION: main
Process: com.rywards, PID: 11477
java.lang.RuntimeException: Camera is being used after Camera.release() was called
    at android.hardware.Camera.lock(Native Method)
    at com.lwansbrough.RCTCamera.RCTCameraModule.releaseMediaRecorder(RCTCameraModule.java:405)
    at com.lwansbrough.RCTCamera.RCTCameraModule.onError(RCTCameraModule.java:130)
    at android.media.MediaRecorder$EventHandler.handleMessage(MediaRecorder.java:1204)
    at android.os.Handler.dispatchMessage(Handler.java:102)
    at android.os.Looper.loop(Looper.java:148)
    at android.app.ActivityThread.main(ActivityThread.java:7406)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
</pre>

gitlovenotwar, 2017-09-14:
Hi, any help on this one? I'm badly need to fix on this one :( Please help.

FarhanKhalid, 2017-09-21:
I can try to help. Can you send me the code that you are using to trigger the startRecording and stopRecording, and both the start and stop recording functions.

gitlovenotwar, 2017-09-25:
Hi, sorry for the late reply. It's something like this.

```
// ... some imports

class Camera extends Component {
  constructor(props) {
    super(props);

    this.state = {
      // ...
    };
  }

  componentDidMount() {
    this._initPanResponder();
  }

  _initPanResponder() {
    this._panResponder = PanResponder.create({
      onStartShouldSetPanResponder: (evt, gestureState) => true,

      onStartShouldSetPanResponderCapture: (evt, gestureState) => true,

      onMoveShouldSetPanResponder: (evt, gestureState) => true,

      onMoveShouldSetPanResponderCapture: (evt, gestureState) => true,

      onPanResponderGrant: (evt, gestureState) => {
        // when pressing again, stop 
        if(this._hasVideoStarts) {
          this._stopVideoRecording();
          return;
        }

        this._hasVideoStarts = true;
        
        // fires to record video
        this._startVideoRecording();
      },

      onPanResponderRelease: (evt, gestureState) => {
        // ... some animations when releasing finger...
      }
    });
  }

  _stopVideoRecording() {

  }

  _startVideoRecording() {
    const options = {
      captureQuality: Camera.constants.CaptureQuality.low,
      totalSeconds: this.state.totalSeconds,
      mode: Camera.constants.CaptureMode.video,
      target: Camera.constants.CaptureTarget.temp
    };

    const onVideoSuccess = (response) => {
      this._stopVideoRecording();
    };

    const onVideoFail = (err) => {
      this._stopVideoRecording();
    };
    
    this.camera.capture(options).then(onVideoSuccess).catch(onVideoFail);
  }

  _stopVideoRecording() {
    if(!this.camera) return;

    this.camera.stopCapture();
  }

  render() {
    return (
      <Camera
        ref={cam => this.camera = cam}
        aspect={Camera.constants.Aspect.fill}
        type={this.state.type}
        flashMode={this.state.flashMode}
        mirrorImage={this.state.mirrorImage}
        captureQuality={this.state.captureQuality}
        orientation={this.state.orientation}
        style={[StyleSheet.absoluteFill, this.state.data ? styles.hide : null]}>

      </Camera>
    )
  }
}
```

FarhanKhalid, 2017-09-26:
Sorry I haven't been able to see anything wrong with your code yet. I thought the crash is being caused by this.camera.stopCapture() being called twice but from your code it doesn't look like that could happen.

n1ru4l, 2018-12-12:
`RCTCamera` is deprecated, please try to use `RNCamera` instead.

closed 2018-12-12 by n1ru4l