react-native-camera/react-native-camera #890: don´t automatically save the picture taken

Lorengamboa, 2017-09-14:
Is it possible to give the user the chance to wether or not save it?

monolithed, 2017-11-06:
I'm absolutely agree, I need a chance to choose where my photo will be located

Lorengamboa, 2017-11-08:
Not also that.
What if you want to manipulate the image and just upload it into a service?

monolithed, 2017-11-08:
Here's an example of my temporary solution:

```ts
import PhotosFramework from 'react-native-photos-framework';
import Camera from 'react-native-camera';

...
constructor (props, context) {
	super(props, context);

	let { Aspect, CaptureTarget, Type, Orientation, FlashMode } = Camera.constants;

	this.state = {
		camera: {
			aspect: Aspect.fill,
			captureTarget: CaptureTarget.cameraRoll,
			onFocusChanged: event => {},
			onZoomChanged: event => {},
			type: Type.back,
			orientation: Orientation.auto,
			fixOrientation: true,
			defaultOnFocusComponent: true,
			mirrorImage: true,
			flashMode: 0,
			...props.options
		}
	};
}

camera: Camera = null;

async saveImage (album: string, uri: string) {
	let { albums: [ TestAlbum ] } = await PhotosFramework.getAlbumsByTitle(album);
	
	if (!TestAlbum) {
		PokerFaceAlbum = await PhotosFramework.createAlbum(album);
	}
	
	let { albums: [ CameraRollAlbum ] } = await PhotosFramework.getAlbums({
		type: 'smartAlbum',
		subType: 'smartAlbumUserLibrary',
		assetCount: 'exact',
	});
	
	await PhotosFramework.createAssets({
		images: [{ uri }],
		album: TestAlbum
	});
}

async getImage () {
	let { path } = await this.camera.capture();

	await saveImage('Test', path);
}
...
```

aamir0, 2018-08-31:
I submitted #1770 to allow for a `doNotSave` flag, which should provide the functionality you're looking for

sibelius, 2018-09-03:
fixed here https://github.com/react-native-community/react-native-camera/pull/1770

test on master

tks

closed 2018-09-03 by sibelius