react-native-camera/react-native-camera #891: How to set metadata correctly and how to retrieve it?

Stophface, 2017-09-15:
I am not sure if I understand the docs properly.
`capture([options]): Promise` allows to set `metadata` like `location`.

However, how do I set the `metadata` correctly?! That way?

```
    _takePhoto() {
        const options = { location };
        this.camera.capture({ metadata: options })
            .then((data) => {
                console.log("photodata", data, data.path)
            })
            .catch(err => console.error(err))
    }
```

And, how would I read the `location` information out again? When I log `console.log("photo data", data, data.path)` there is no `location` information in it.

koswarabilly, 2017-11-17:
i got the same problem , mine even crash when i try to take photo

zahid-dapperapps, 2017-12-05:
+1 same issue no metadata being saved

closed 2019-06-08 by stale[bot]