react-native-camera/react-native-camera #892: Get base64 encoded string

Stophface, 2017-09-15:
I wonder how I get the `base64` encoded string. `captureTarget` can be set to `Camera.constants.CaptureTarget.memory` which should return the `base64` encoded string. Although it mentions that is deprecated, the docs say it returns the `base64` encoded string. 

> When using the deprecated memory output, the image binary is sent back as a base64-encoded string.

However, logging the data, returns me a link to a location:

```
    _takePhoto() {
        const options = {};
        this.camera.capture({ metadata: options })
            .then((data) => {
                console.log("photo data", data, data.path)
            })
            .catch(err => console.error(err))
    }
```

Is `base64` not supported anymore?

augustolazaro, 2017-12-05:
I think base64 is at `data.data`

sibelius, 2017-12-10:
@augustolazaro answer is correctly

closed 2017-12-10 by sibelius

pacozaa, 2018-01-24:
@augustolazaro  @sibelius my `data.data` is `undefined`