react-native-camera/react-native-camera #893: App crash on Kitkat

RZulfikri, 2017-09-15:
Hi everyone, i got an error while using react-native-camera, but this error only show when i run in device Xiaomi Redmi 1S with OS:Kitkat. And it work fine when i'm using Emulator Device (Nexus 5 with OS Oreo) and Other Devices that use OS Marshmellow and Nougat

Environtment :
react-native 4.2
react-native-camera 0.10

this the error log.

```
09-15 16:39:31.302 23015-23015/com.sweatcircleapp E/AndroidRuntime: FATAL EXCEPTION: main
                                                                    Process: com.sweatcircleapp, PID: 23015
                                                                    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.sweatcircleapp/com.sweatcircleapp.MainActivity}: java.lang.RuntimeException: Requested enabled DevSupportManager, but DevSupportManagerImpl class was not found or could not be created
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2200)
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2249)
                                                                        at android.app.ActivityThread.access$800(ActivityThread.java:141)
                                                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1212)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                        at android.os.Looper.loop(Looper.java:136)
                                                                        at android.app.ActivityThread.main(ActivityThread.java:5052)
                                                                        at java.lang.reflect.Method.invokeNative(Native Method)
                                                                        at java.lang.reflect.Method.invoke(Method.java:515)
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:796)
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:612)
                                                                        at dalvik.system.NativeStart.main(Native Method)
                                                                     Caused by: java.lang.RuntimeException: Requested enabled DevSupportManager, but DevSupportManagerImpl class was not found or could not be created
                                                                        at com.facebook.react.devsupport.DevSupportManagerFactory.create(DevSupportManagerFactory.java:77)
                                                                        at com.facebook.react.ReactInstanceManager.<init>(ReactInstanceManager.java:326)
                                                                        at com.facebook.react.ReactInstanceManagerBuilder.build(ReactInstanceManagerBuilder.java:227)
                                                                        at com.facebook.react.ReactNativeHost.createReactInstanceManager(ReactNativeHost.java:84)
                                                                        at com.facebook.react.ReactNativeHost.getReactInstanceManager(ReactNativeHost.java:41)
                                                                        at com.facebook.react.ReactActivityDelegate.loadApp(ReactActivityDelegate.java:111)
                                                                        at com.facebook.react.ReactActivityDelegate.onCreate(ReactActivityDelegate.java:100)
                                                                        at com.facebook.react.ReactActivity.onCreate(ReactActivity.java:54)
                                                                        at com.sweatcircleapp.MainActivity.onCreate(MainActivity.java:16)
                                                                        at android.app.Activity.performCreate(Activity.java:5242)
                                                                        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087)
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2164)
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2249) 
                                                                        at android.app.ActivityThread.access$800(ActivityThread.java:141) 
                                                                        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1212) 
                                                                        at android.os.Handler.dispatchMessage(Handler.java:102) 
                                                                        at android.os.Looper.loop(Looper.java:136) 
                                                                        at android.app.ActivityThread.main(ActivityThread.java:5052) 
                                                                        at java.lang.reflect.Method.invokeNative(Native Method) 
                                                                        at java.lang.reflect.Method.invoke(Method.java:515) 
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:796) 
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:612) 
                                                                        at dalvik.system.NativeStart.main(Native Method) 
                                                                     Caused by: java.lang.reflect.InvocationTargetException
                                                                        at java.lang.reflect.Constructor.constructNative(Native Method)
                                                                        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
                                                                        at com.facebook.react.devsupport.DevSupportManagerFactory.create(DevSupportManagerFactory.java:70)
                                                                        at com.facebook.react.ReactInstanceManager.<init>(ReactInstanceManager.java:326) 
                                                                        at com.facebook.react.ReactInstanceManagerBuilder.build(ReactInstanceManagerBuilder.java:227) 
                                                                        at com.facebook.react.ReactNativeHost.createReactInstanceManager(ReactNativeHost.java:84) 
                                                                        at com.facebook.react.ReactNativeHost.getReactInstanceManager(ReactNativeHost.java:41) 
                                                                        at com.facebook.react.ReactActivityDelegate.loadApp(ReactActivityDelegate.java:111) 
                                                                        at com.facebook.react.ReactActivityDelegate.onCreate(ReactActivityDelegate.java:100) 
                                                                        at com.facebook.react.ReactActivity.onCreate(ReactActivity.java:54) 
                                                                        at com.sweatcircleapp.MainActivity.onCreate(MainActivity.java:16) 
                                                                        at android.app.Activity.performCreate(Activity.java:5242) 
                                                                        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1087) 
                                                                        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2164) 
                                                                        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2249) 
[... cut ...]
                                                                        at java.lang.reflect.Method.invoke(Method.java:515) 
                                                                        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:796) 
                                                                        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:612) 
                                                                        at dalvik.system.NativeStart.main(Native Method) 

```

anyone know what should i do ?

thanks.

ngnclht1102, 2017-10-17:
I have the same problem on Kitkat too

ngnclht1102, 2017-10-18:
After 1day dealing with that kind of error, I figure out that the error is occured because I've enable  multidex but I used wrong Application class. See more: https://developer.android.com/studio/build/multidex.html

jslok, 2017-10-21:
@ngnclht1102 Thank you for the solution. I needed some additional searching to then find out I need to import multidex module in MainApplication.java, but couldn't have figured it out without you.

For anybody else that has this issue, the solution is to edit your MainApplication.java to extend MultiDexApplication like in the link above, but also import import android.support.multidex.MultiDexApplication; in this file.

closed 2018-09-23 by SimonErm

yasinugrl, 2019-03-03:
Hi @ngnclht1102 

Please show how to use and which Application class use for solve that problem?

RZulfikri, 2019-03-03:
@yasinugrl check this https://developer.android.com/studio/build/multidex#mdex-gradle