react-native-camera/react-native-camera #894: TypeError: unsupported file type: undefined

007agentHP, 2017-09-15:
Hello, so i'm trying to just place an image on screen with the camera in the background. 

```
import React, { Component } from 'react';
import { View, Image } from 'react-native';
import Camera from 'react-native-camera';

class CameraView extends Component {
  takePhoto() {
    const options = {};
    this.refs.cam.capture({ metadata: options })
      .then((data) => console.log(data))
      .catch(err => console.error(err));
  }
  render() {
    return (
      <View style={styles.containerStyle}>
        <Camera
          style={styles.previewStyle}
          ref={'cam'}
        >
          <Image source={require('../images/cameraclicker.png')} />
        </Camera>
      </View>
    );
  }
}

const styles = {
  containerStyle: {
    flex: 1,
    flexDirection: 'row'
  },
  previewStyle: {
  flex: 1,
  justifyContent: 'flex-end',
  alignItems: 'center'
  },
  captureStyle: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    color: '#000',
    padding: 10,
    margin: 40
  }
};

export default CameraView;
```

and no matter what i do, it throws the error:
```
Loading dependency graph, done.
Bundling `index.ios.js`  [development, non-minified, hmr disabled]  100.0% (350/350), failed.
TypeError: unsupported file type: undefined (file: /Users/007agenthp/Desktop/Programming/React-Native/Workspace/SM_WIP/images/cameraclicker.png)
    at lookup (/Users/007agenthp/Desktop/Programming/React-Native/Workspace/SM_WIP/node_modules/image-size/lib/index.js:34:9)
    at /Users/007agenthp/Desktop/Programming/React-Native/Workspace/SM_WIP/node_modules/image-size/lib/index.js:93:22
    at /Users/007agenthp/Desktop/Programming/React-Native/Workspace/SM_WIP/node_modules/image-size/lib/index.js:50:9
    at /Users/007agenthp/Desktop/Programming/React-Native/Workspace/SM_WIP/node_modules/graceful-fs/graceful-fs.js:43:10
    at FSReqWrap.oncomplete (fs.js:135:15)
```

Fconstant, 2018-02-27:
Your code seems ok to me. This seems like a module resolving issue rather than this `react-native-camera`'s problem.
What's your react-native version?

Try moving your `Image` outside the `Camera`, with `position: absolute` to see if it works out. Althought I don't believe that's the case.

Also, Try to require your image outside the component and store it as a `const`.

sibelius, 2018-02-27:
Not a problem with this package

closed 2018-02-27 by sibelius