react-native-camera/react-native-camera #895: Camera crash when I tried to open it again

nachovg, 2017-09-16:
Hi, thanks for your attention! I'm developing an app to scan qr codes and I have a little problem, when I scan a Qr I go to a success or fail screen, when is fail I can try it again but the app crashes when the scan screen render the camera the second time. I think it happens cause the camera is running in background, there is any way to close it completely?

satishsynnov, 2017-09-26:
Can you please share your crash report?

nachovg, 2017-09-26:
There is no crash report, it just close it

satishsynnov, 2017-09-27:
Can you share your code how are you using?

qamoonsite, 2018-07-24:
happens to me also, please help

n1ru4l, 2018-12-19:
Please open a new issue if you are still experiencing this problem with the newest version of `RNCamera`.

closed 2018-12-19 by n1ru4l