react-native-camera/react-native-camera #896: Undefined is not an object (evaluating 'CameraManager.Aspect') during running app

AAGSICON, 2017-09-19:
[image]

**# App.js**

import React from 'react';
import {Text, View} from 'react-native';
 import Registration  from './src/pages/Registration.js';
         // sample styles

export default class App extends React.Component {
  render() {
    return (
      <Registration/>

    );
  }
}

# **Registration.js**

import React, { Component } from 'react';
import {  Text, TextInput, View, Button, Form, NativeAppEventEmitter } from 'react-native';
import Camera from './react-native-camera/index.js';

export default class Registration extends Component {

constructor(props) {
    super(props);
    this.state = {text: ''};
  }
  

render() {
        return (
          <View style={{padding: 30}}>
            <TextInput
             style={{height: 40}}
              autoFocus={true}
              autoCapitalize="none"
              autoCorrect={false}
              placeholder="First Name"
              onChangeText={(firstName) => this.setState({firstName})}
            />

          <TextInput
           style={{height: 40}}
            autoFocus={false}
            autoCapitalize="none"
            autoCorrect={false}
            placeholder="Last Name"
            onChangeText={(lastName) => this.setState({lastName})}
          />
          <TextInput
           style={{height: 40}}
           autoFocus={false}
           autoCapitalize="none"
           autoCorrect={false}
           placeholder="Email"
           onChangeText={(email) => this.setState({email})}
         />

          <TextInput
          style={{height: 40}}
          placeholder="Username!"
          onChangeText={(text) => this.setState({text})}
        />

	       <TextInput
          style={{height: 40}}
          placeholder="password!"
          onChangeText={(text) => this.setState({text})}
        />

          <TextInput
           style={{height: 40}}
            secureTextEntry={true}
            autoCapitalize="none"
            autoCorrect={false}
            placeholder="Password Confirmation"
            onChangeText={(passwordConfirmation) => this.setState({passwordConfirmation})}
          />
          <Camera
           ref={(cam) => {
             this.camera = cam;
           }}
           style={styles.preview}
           aspect={Camera.constants.Aspect.fill}>
           <Text style={styles.capture} onPress={this.takePicture.bind(this)}>[CAPTURE]</Text>
       </Camera>
        <Button
 		 onPress={this.camera.capture().then((data) => console.log(data)).catch(err => console.error(err))}
		  title="Press Me"
	/>
      </View>

        );
    }
  }

ecavalcanti, 2017-09-20:
Add the following permissions in the AndroidManifest.xml file:
```
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
```

AAGSICON, 2017-09-20:
Already added

ashishrokr, 2017-09-20:
Please Add this into your gradle file

android {
...
defaultConfig {
    ...
    multiDexEnabled true
    }
}
AND also add the below dependency in your gradle

dependencies {
  compile 'com.android.support:multidex:1.0.1'
}  
This will definitely solve your issue.

AAGSICON, 2017-09-20:
In reactnative project folder where is it

ashishrokr, 2017-09-20:
in Android/App/build.gradle

AAGSICON, 2017-09-20:
in reactnative project there is 2 folders that given below in which folder Android/App/build.gradle

react-native-packager-cache-38539b997296799ed5a5b31713a7ada56b73934e
node_modules

cvng, 2017-09-23:
Solution that works for me (on ios) : run the app via Xcode directly instead of `react-native run-ios`

tlemesle, 2017-09-29:
I have the same problem. I have already setup my project with the manual install process, rnpm link and @ashishrokr solution, but nothing works. Someone have another solution ?

Test with a Galaxy S7Edge on Android 7.0
react-native-cli: 2.0.1
react-native: 0.47.2
expo: 44.0.0

haikezegwaard, 2017-10-14:
Same problem here, anyone solved this for `npm run ios`?

moksamedia, 2017-10-16:
@haikezegwaard - Are you running in the simulator? I get this error when I use the <Camera/> in the iOS simulator but it works on my device. The iOS simulator doesn't support the camera, so my guess is that the library isn't failing very gracefully.

FacuAcosta, 2017-11-07:
I don't know if you still have this issue but I've been dealing with it the whole morning and I guess I fixed It. As the docs says:

> On Android, you require buildToolsVersion of 25.0.2+. This should easily and automatically be downloaded by Android Studio's SDK Manager.

So go to your `android/app/build.gradle` and make sure that `buildToolsVersion` is at least 25.0.2. Mine was on 23.0.1 so after changing this and updating all the tools I had on Android Studio the problem was fixed.

AAGSICON, 2017-11-08:
how can I acess camera in react native webview?

JimmyMow, 2017-11-19:
I am seeing this on iOS. Any solution?

ashishrokr, 2017-11-20:
in IOS you have given Camera permission in info.plist ?

dvisco, 2018-01-15:
Make sure you run `pod install` after linking the project via `react-native link`. That's what had been giving me this error.

jgcmarins, 2018-01-15:
This was fixed on `master`. Could you try on `master branch`? Thanks.
https://github.com/react-native-community/react-native-camera#how-to-use-master-branch

s349856186, 2018-01-18:
@jgcmarins I had try, but not fix

ashishrokr, 2018-01-18:
@s349856186  you are on which react-native version ?Provide your Package.json for further help

s349856186, 2018-01-18:
"react": "16.0.0",
    "react-native": "0.51.0",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",

@ashishrokr

ashishrokr, 2018-01-18:
okay, your build was successfull ? any errors facing in building the project?

s349856186, 2018-01-18:
yes，** BUILD SUCCEEDED **
@ashishrokr

ryd0rz, 2018-01-18:
Hmm. Not working for me either:

    "react": "16.0.0-alpha.12",
    "react-native": "0.51.0",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",

s349856186, 2018-01-19:
@ashishrokr Thanks for your guidance.
rnpm-install info Linking react-native-camera ios dependency 
rnpm-install info iOS module react-native-camera has been successfully linked 

<key>NSPhotoLibraryAddUsageDescription</key>
    <string>Your message to user when the camera is accessed for the first time</string>
    <key>NSLocationWhenInUseUsageDescription</key>
    <string></string>
    <key>NSMicrophoneUsageDescription</key>
    	<string>This app needs access to the microphone to take videos.</string>
    <key>NSCameraUsageDescription</key>
    <string>Your message to user when the camera is accessed for the first time</string>
    <key>NSPhotoLibraryUsageDescription</key>
    <string>Your message to user when the photo library is accessed for the first time</string>
    <key>NSMicrophoneUsageDescription</key>
    <string>Your message to user when the microphone is accessed for the first time</string>
Please help me to check if its correct?

egunsoma, 2018-01-22:
neither for me:
```
    "react": "16.2.0",
    "react-native": "0.52.1",
    "react-native-firebase": "^3.2.2",
    "react-native-camera": "0.13.0"
```

sibelius, 2018-01-23:
Make sure the link is correct

closed 2018-01-23 by sibelius

juicycleff, 2018-01-23:
This issue never ever get fixed

egunsoma, 2018-01-23:
@sibelius, how can I check if the linking works? what should I see and where?

sibelius, 2018-01-23:
```
import { NativeModules } from 'react-native'
console.log(NativeModules)
```

egunsoma, 2018-01-23:
@sibelius My question was wrong. When I'm logging the NativeModules, the Camera is not present. How can I fix this? I run react-native link, and I have Cocoapods. If I dont run pod install, it compiles but throws this error. If I run pod install, it throws ambigous React and all other component defintions error.

sibelius, 2018-01-23:
check cocoapods docs, if the docs is wrong, we should improve it

can u do that?

egunsoma, 2018-01-23:
After running pod install, I get this error:
[image]

egunsoma, 2018-01-23:
@sibelius Any idea why?

egunsoma, 2018-01-23:
@sibelius After running yarn start --reset-cache, this is how my packager looks like:
[image]

sibelius, 2018-01-23:
@jgfidelis can u help on this?

ishigamii, 2018-02-20:
Same problem the pod install has a dependency to react that makes this. 
Any solution ?

sibelius, 2018-02-20:
@bkvirendra  could help on `pod` installs

ishigamii, 2018-02-24:
Still stuck with this. Anything to resolve this ?

bkvirendra, 2018-02-24:
@ishigamii This was resolved in https://github.com/react-native-community/react-native-camera/commit/8989aa235475e3157ca3b172632fdbd26515f52a

Can you point your npm installation to use the latest master instead as this was merged recently but isn't yet included in a release? Like

`"react-native-camera": "https://github.com/react-native-community/react-native-camera/tarball/master"`

ishigamii, 2018-02-26:
@bkvirendra actually just did what you told me and doing so I got a clean install but got back to the "CameraManager.aspect" Error. So I linked the project `react-native link react-native-camera` and `pod install` and the pod install installed React again and back to beginning of the problem :(

bkvirendra, 2018-02-26:
@ishigamii Are you using cocoapods for all the RN libraries & could you also share the environment details of your project?

The `react-native-camera` does work with the simulator as well, as long the library is linked properly.

ishigamii, 2018-02-26:
@bkvirendra I don't use pods for all of them, but only the one requiring it. 
Here is the details : 

Pods : 
  pod 'Firebase/Core'
  pod 'Firebase/Auth'
  pod 'Firebase/Database'
  pod 'Google/SignIn'
  pod 'react-native-camera', :path => '../node_modules/react-native-camera'

Package.json : 
  "dependencies": {
    "debounce": "^1.1.0",
    "moment": "^2.20.1",
    "moment-timezone": "^0.5.14",
    "react": "16.2.0",
    "react-moment": "^0.7.0",
    "react-native": "0.53.0",
    "react-native-elements": "^0.19.0",
    "react-native-firebase": "^3.2.4",
    "react-native-google-signin": "^0.12.0",
    "react-native-i18n": "^2.0.6",
    "react-native-vector-icons": "^4.5.0",
    "react-navigation": "^1.0.0-beta.27",
    "react-native-camera": "https://github.com/react-native-community/react-native-camera/tarball/master"
  },

React : 
   react-native-cli: 2.0.1
   react-native: 0.53.0

Right now, if I remove the pod from react-native-camera It works (after some cache cleaning), if I add the pod it install React (like described here: https://github.com/react-native-community/react-native-camera/issues/1237). If I don't pod install I get the aspect problem. 

Thanks a lot for you help :)

ishigamii, 2018-02-28:
Still blocked in ios by this :s Do you need more information ?

dann1609, 2018-03-19:
do a manual link:

1) npm install react-native-camera --save
2) In XCode, in the project navigator, right click Libraries ➜ Add Files to [your project's name]
3) Go to node_modules ➜ react-native-camera and add RNCamera.xcodeproj
4) In XCode, in the project navigator, select your project. Add libRNCamera.a to your project's Build Phases ➜ Link Binary With Libraries
5) Click RNCamera.xcodeproj in the project navigator and go the Build Settings tab. Make sure 'All' is toggled on (instead of 'Basic'). In the Search Paths section, look for Header Search Paths and make sure it contains both $(SRCROOT)/../../react-native/React and $(SRCROOT)/../../../React - mark both as recursive.

FlaviooLima, 2018-04-16:
Try doing [this](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md)