react-native-camera/react-native-camera #900: Crash on Android 4.4.4 every time

novellizator, 2017-09-21:
Whenever we try to take a picture (in Samsung Galaxy S2), the whole app crashes.

cloudsafe, 2017-09-25:
the same

sahir, 2017-09-27:
can you Please provide crash logs.

cloudsafe, 2017-09-27:
@sahir I found the reason.
Configure Apps with Over 64K Methods.
And I was configure error.
Now,I want display status bar ,when I use camera .
Do you have some ideas?

novellizator, 2017-09-27:
Ok solved, https://github.com/lwansbrough/react-native-camera/issues/590 is the issue. On android I had to set up a larger heap.

closed 2017-09-27 by novellizator

sahir, 2017-09-27:
@cloudsafe you can customize Camera view using **styles.overlay, styles.topOverlay** for example in my case I add Button and text in Camera View: https://github.com/sahir/react-native-barcode-reader

cloudsafe, 2017-10-05:
@sahir sorry,I was said it wrong,I want hidden the status bar. now I finished .
set 
_viewFinder.setSystemUiVisibility(SYSTEM_UI_FLAG_FULLSCREEN);
in RCTCameraView.java