react-native-camera/react-native-camera #905: Android compiling error

fenicento, 2017-09-25:
I have various errors when i run react-native run-android, mostly referring to non existing libraries:

error: package com.drew.imaging does not exist
error: package com.drew.metadata does not exist
error: package com.google.zxing does not exist

should I install them on my own? should they be included in google APIs?

fenicento, 2017-09-25:
nevermind, I didn't update build.gradle. I have another error now:

Execution failed for task ':app:transformClassesWithDexForDebug'.
> com.android.build.api.transform.TransformException: com.android.ide.common.process.ProcessException: java.util.concurrent.ExecutionException: com.android.dex.DexException: Multiple de
x files define Lcom/lwansbrough/RCTCamera/RCTCamera$CameraInfoWrapper;

any clues?

samhunta, 2017-10-03:
Try removing specific versions from your build.gradle

Source: https://github.com/tolu360/react-native-google-places/issues/31

n1ru4l, 2018-09-25:
Dpulicate of #885

closed 2018-09-25 by n1ru4l