react-native-camera/react-native-camera #908: [Discussion] Ways to make capture faster?

wassgha, 2017-09-26:
Hello everyone,
I opened this thread to discuss implementation tricks / ways to improve `react-native-camera` in order to get faster capture.
I have noticed that the `capture` promise only resolves after a second or two. During the time after the capture button is clicked and before the photo appears, the preview window stops for an instant then resumes showing a blurry image (because of user's movement) then my app navigates to the preview screen then the photo is loaded. This makes a really bad experience for the user compared to other react native apps such as Facebook Messenger or Instagram whose capture function is instantaneous.

#### Possible sources of slowdown:
* Camera seems to adjust focus most of the time after the capture button is clicked. In other react native apps the camera simply captures what's on the preview instantly without any focus adjustment
* Image post-processing? (haven't looked into the code)
* Preview window keeps running while switching to the captured image page?
#### What I have tried (on Android) :
* Decreasing the capture quality (did not have much effect until Quality.LOW which improved capture time but the quality was too bad to be useful)
* Switching from saving to disk to saving to temp (didn't help)
* Switching to Quality.preview (didn't notice any change)

Do you have any suggestions or any implementations of `react-native-camera` that have the same speed as Messenger/Instagram? Do you have any ideas to improve the codebase for this problem? Please share!

Thank you.

ippa, 2017-09-30:
I just want to add I've experinced the same thing the last year with our app.

wassgha, 2017-10-11:
@ippa have you solved the problem?

6pm, 2017-10-24:
I set captureQuality="medium" and image size is 170-220kb.
By default this option is high and size  0.9-1Mb.

wassgha, 2017-10-24:
I ended up making it synchronous. My app doesn't require a preview screen
after the photo is taken so pressing the button would display an icon
indicating that something is happening in the background while the app is
waiting for the photo data to come through, resize it using another module
then upload it to AWS.

Cygnus2112, 2017-10-26:
In my experience, the delay seems to be much longer for Android. Didn't even notice it in iOS, really.

Cygnus2112, 2017-10-26:
Also, the delay seem to be the longest for the first photo taken after the component mounts. Capture speed seems much faster for subsequent photos.

wassgha, 2017-11-17:
Hey everyone, just a quick update. I found out that my problem was mainly caused by the redux logger, after I removed that from the middleware the app became way faster. Just one thing to add to the list!

6pm, 2017-11-17:
@wassgha It is common problem with console.log in react-native

wassgha, 2017-11-30:
@all the best solution that I found (works perfectly for me, probably not for other uses) is to use  [`react-native-view-shot`](https://github.com/gre/react-native-view-shot) to capture the actual perview window. For me at least, this solves a lot of problems including speed, image size/compression and the front-end camera that used to be reversed. Again, this might not be appropriate for every application (if you need higher quality images or a different aspect ratio).

truepic-tech, 2017-12-19:
@wassgha Can you provide a code snippet or elaborate on how you're using react-native-view-shot to capture the preview image from react-native-camera?  All of my attempts result in an empty black image.

wassgha, 2017-12-19:
@truepic-tech I'm actually getting the exact same problem on iOS and just started debugging it. It works perfectly on Android but I haven't tried it on iOS till now. Can you confirm that you're using iOS?

timLoewel, 2017-12-20:
This works for me in android:
`    
import { captureRef } from 'react-native-view-shot';
...
captureRef(this.camera, this.screenCaptureOptions)
      .then(uri => {
        this.props.currentSite && this.props.setSite(this.props.currentSite);
        this.props.createPhotoPreview(uri);
        this.setState({ capturing: false });
        this.navigateToNextScreen();
      })`

adhenrique, 2018-01-30:
[https://github.com/react-native-community/react-native-camera/issues/908](https://github.com/react-native-community/react-native-camera/issues/908)