react-native-camera/react-native-camera #913: Background color of camera

kevinresol, 2017-09-28:
When `aspect` is set to `fit`, there would be some extra padding in black color for the longer dimension. I am interested in setting the color of such padding. Tried setting `backgroundColor` in camera's style but doesn't work. Any idea?

tuluffy, 2017-11-17:
I'm get this problem, do u solve it?

kevinresol, 2017-11-17:
@tuluffy  see the linked PR

nihp, 2019-10-28:
I am facing this issue now. Any solution for this

```
<Camera
          ref={(cam) => {
            this.camera = cam
          }}
          style={{flex: 0,
    justifyContent: 'flex-end',
    alignItems: 'center',
    backgroundColor: 'red',
    width: deviceWidth,
    height: deviceWidth * (5/4.2)}}
          >
        </Camera>
```