react-native-camera/react-native-camera #917: Unable to read Code 128 type barcodes

connercms, 2017-09-29:
I am trying to read a Code 128 type barcode, but the onBarCodeRead event is never triggered. Here is a sample of the barcode.

[image]

The following is my code 

<Camera style={{ flex: 1 }} onBarCodeRead={e => {
    this.setState({ showCamera: false })
    Alert.alert("BarCode Read", "Data: " + JSON.stringify(e, null, 4), [{ text: "OK", onPress: () => null }], {
      cancelable: true
    })
  }} />
  }

Using this i can successfully read and show an alert on QR codes. I see there are several issues open for Code 128 read issues, but have seen no resolution.

muhammadnasir786, 2018-04-03:
same problems here !

closed 2019-06-08 by stale[bot]

MinhHieu6796, 2020-07-29:
same problems here ! anybody help this problem

R4DIC4L, 2021-01-19:
I tested your barcode and it works on my side with `"react-native-camera": "3.40.0"`. You must check `event.data`, `event` is an object with other members, see below typescript definition.

```typescript
interface BarCodeReadEvent {
  data: string;
  rawData?: string;
  type: keyof BarCodeType;
  /**
   * @description For Android use `{ width: number, height: number, origin: Array<Point<string>> }`
   * @description For iOS use `{ origin: Point<string>, size: Size<string> }`
   */
  bounds:
    | { width: number; height: number; origin: Array<Point<string>> }
    | { origin: Point<string>; size: Size<string> };
  /**
   * Raw image bytes in JPEG format (quality 100) as Base64-encoded string, only provided if `detectedImageInEvent=true`.
   */
  image: string;
}
```

I am also using `"@nartc/react-native-barcode-mask": "1.2.0"` to display a scanner line, but it is not required, see below sample code:

```typescript
import { BarCodeReadEvent, RNCamera } from 'react-native-camera';
import { BarcodeMask } from '@nartc/react-native-barcode-mask';
...
barcodeDetected = (event: BarCodeReadEvent) => {
  const { handleBarcodeDetected } = this.props;
  if (handleBarcodeDetected) {
    handleBarcodeDetected(event.data);
  }
}

render() {
  return (
    <RNCamera
      type={RNCamera.Constants.Type.back}
      captureAudio={false}
      onBarCodeRead={this.barcodeDetected}
    >
      <BarcodeMask animatedLineColor={'red'} />
    </RNCamera>
  );
```