react-native-camera/react-native-camera #918: App Crash when taking a picture

sandropoluan, 2017-09-29:
App get crashed/unexpected closed after taking a picture, And no image was saved.
This problem only happen with real device. Everything is fine on emulator. 

Everything was fine when i was using RN 0.44.0, but the problem appear after upgrading to  RN 0.48.4.

react-native-camera@0.10.0
React Native@0.48.4

Device
Xiaomi Redmi Note 2 (Android 5.0.2)

goel135, 2017-10-25:
+10000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

goel135, 2017-10-25:
Any update on this because many people are facing this issue.
If there is any update do reply asap.
I have tried it on react-native-camera@0.6 as well as on react-native-camera@0.10
With react native version 0.46.4
and react navigation version ^1.0.0-beta.11
Do reply if there is any solution.

koswarabilly, 2017-11-17:
i am facing the same issue but only when i try to add location metadata to it

seanelias, 2018-02-28:
Same issue app crashes when taking a photo. I tested this on Pixel, Pixel 2 XL running oreo.

sibelius, 2018-02-28:
Use master and RNCamera 

Also provide stacktrace of the error

closed 2018-02-28 by sibelius