react-native-camera/react-native-camera #920: Turn Off and On Camera

joecaraccio, 2017-09-30:
Sorry if this isn't an issue.. Couldn't find a good way to ask this, but is there a way to turn on and off the camera? My on the fly solution was to just do something like:

render(){
   if(this.state.cameraOn == true){
     <Camera/>
   }
}

Does pulling the camera out of the render loop 'turn it off'. Obviously using the camera is expensive power wise.. so when my users aren't using it I would like to be able to shut it down.

Any help would be huge!

Onumis, 2017-10-02:
I use an approach similar to this this one. To ensure camera is disabled when navigating to another screen/tab.

wysiecki, 2017-10-06:
Hi, but how can i disable the cam ?

thx

Onumis, 2017-10-06:
@wysiecki something like this, in my camera screen:

```

  render() {
    if (this.props.cameraActive) {
      return this.renderCamera()
    } else {
      return null
    }
  }

renderCamera() {
    return (
      <View style={
        [
          viewStyles.container,
          this.props.style
        ]
      }>
        <Camera
          style={...}
          aspect={Camera.constants.Aspect.fill}
          barcodeTypes={
            (Platform.OS === 'android') ? ['qr'] : ['org.iso.QRCode']
          }
          onBarCodeRead={...}>
        </Camera>
      </View>
    )
  }
}
```

Basically, whenever I'm navigating away from this screen, I set this to false, and when navigating into this screen, I set to true. You can do this only on the `componenDidMount` function, or by passing the control var `cameraActive` as a property (like I do here), or use the state, Redux, etc.

Titinious, 2018-02-21:
I think you could do this instead as well. Works for me :) 

```
render(){
  return(
    <View>
      {this.state.isCamOn && <Camera>
      </Camera>}
    </View>
  );
}
```

azizazlan, 2018-02-27:
Hi @Onumis , I followed your suggestion. My react-navigation and redux works. In fact my RNCamera works and it can scan qrcode too. However the camera still on when I switch to another tab.

The only diff with your above code is **this.props.navigation.state.routeName===''Verify'** line instead of "this.props.cameraActive". Somehow I cannot get access **shouldRenderCamera** as I declared in the nav.js file.

Verify.js
```
import { Scanner } from '../components/Scanner';
class Verify extends Component {
  render() {
    if (this.props.navigation.state.routeName === 'Verify') {
      return <Scanner />;
    }
    return null;
  }
}
```
/reducers/index.js

```
export default combineReducers({
  nav,
...
});
```

reducers/nav.js
```
const initialState = {
  ...TabNavigator.router.getStateForAction(NavigationActions.init()),
  shouldRenderCamera: false,
};

export default (state = initialState, action) => {
  // console.log(`Initial state shouldRenderCamera: ${state.shouldRenderCamera}`);
  const nextState = {
    ...TabNavigator.router.getStateForAction(action, state),
    shouldRenderCamera: action.routeName === 'Verify',
  };
  // console.log(`... state shouldRenderCamera: ${nextState.shouldRenderCamera}`);
  return nextState || state;
};
```
index.js
```
import Navigator from './config/routes';

const App = props => (
  <Navigator
    navigation={addNavigationHelpers({
      dispatch: props.dispatch,
      state: props.nav,
      addListener,
    })}
  />
);

const mapStateToProps = state => ({
  nav: state.nav,
  navigationState: state.tabBar,
  shouldRenderCamera: state.shouldRenderCamera,
});

const AppWithReduxNav = connect(mapStateToProps)(App);

export default () => (
  <Provider store={store}>
    <AppWithReduxNav />
  </Provider>
);

```

./config/routes

```
export default TabNavigator(
  {
    Home: {
      screen: Home,
    },
    Verify: {
      screen: Verify,
    },
  },
...
...
)
```

Onumis, 2018-02-27:
@azizazlan I'm not sure if `props.navigation.state` gets updated when you navigate away.
Remember that each screen _runs separately_... meaning that the new screen will have a new `props.navigation.state` but the previous one, most likely, will have the same.

As long as the previous screen remains in memory, the camera will be active.
The `TabNavitagor` will keep adjacents screens in memory to improve response times.

So, my advise is to control camera state via redux, or something else other than props like `navigation`.

matamicen, 2018-08-05:
Hi guys,
We are having a crash in Android when we rotate the emulator when the camera is active, could be the react-navigator with redux? becasue we have the same project with the same versions of react-native-camera, compileSdkVersion, buildToolsVersion, etc and it is working without any problem (the working project doesn't have redux and react-navigation). 

More info: The camera is on a StackNavigator Screen 

it throws this error:
java.io.IOException: setPreviewTexture failed

setPreviewTexture failed
setUpPreview
    Camera1.java:176
onSurfaceChanged
    Camera1.java:109
dispatchSurfaceChanged
    PreviewImpl.java:56
onSurfaceTextureSizeChanged
    TextureViewPreview.java:53
onSizeChanged
    TextureView.java:366
sizeChange
    View.java:17641
setFrame
    View.java:17603
layout
    View.java:17520
onLayout
    RNCameraView.java:191
layout
    View.java:17523
layout
    ViewGroup.java:5612
updateLayout
    NativeViewHierarchyManager.java:241
updateLayout
    NativeViewHierarchyManager.java:206
execute
    UIViewOperationQueue.java:175
run
    UIViewOperationQueue.java:894
flushPendingBatches
    UIViewOperationQueue.java:1001
access$2400
    UIViewOperationQueue.java:46
doFrameGuarded
    UIViewOperationQueue.java:1061
doFrame
    GuardedFrameCallback.java:29
doFrame
    ReactChoreographer.java:134
doFrame
    ChoreographerCompat.java:105
run
    Choreographer.java:869
doCallbacks
    Choreographer.java:683
doFrame
    Choreographer.java:616
run
    Choreographer.java:857
handleCallback
    Handler.java:751
dispatchMessage
    Handler.java:95
loop
    Looper.java:154
main
    ActivityThread.java:6077
invoke
    Method.java
run
    ZygoteInit.java:866
main
    ZygoteInit.java:756

n1ru4l, 2018-12-19:
Seems like this issue is resolved. If you experience any issues I recommend testing if the issue is still occurring with the newest version of `RNCamera`.

closed 2018-12-19 by n1ru4l