react-native-camera/react-native-camera #921: tests fail to: Cannot read property 'default' of undefined after mocking the react-native-camera

niom, 2017-10-01:
When I have the react-native-camera used inside my project my tests will fail with out mocking the react-native-camera:
● Test suite failed to run (yarn test)

    TypeError: Cannot read property 'Aspect' of undefined

      at Object.<anonymous> (node_modules/react-native-camera/index.js:259:78)
      at Object.<anonymous> (index.android.js:14:24)
      at Object.<anonymous> (__tests__/index.android.js:3:19)

and after mocking (yarn test):

 ● Test suite failed to run 

    TypeError: Cannot read property 'default' of undefined

      at Object.user:~/jesttestapp/__mocks__/react-native-camera.js: (__tests__/index.android.js:8:70)
      at Object.<anonymous> (index.android.js:15:24)
      at Object.<anonymous> (__tests__/index.android.js:8:195)

I have created a vanilla react-native application by running:
```
react-native init jesttestapp 
```

Added the react-native-camera to the android side by following the instructions on the README by using:

````
react-native link react-native-camera
````
created a mock for react-native-camera under __mocks_ folder:
````
 
//__mocks__/react-native-camera.js
import React from 'react';

const constants = constants = {
    Aspect: {},
    BarCodeType: {},
    Type: {},
    CaptureMode: {},
    CaptureTarget: {},
    CaptureQuality: {},
    Orientation: {},
    FlashMode: {},
    TorchMode: {}
};

class Camera extends React.Component {

    static constants = constants
    render() {
        return null;
    }
}

Camera.constants = constants;

module.exports = {default: constants};

````
and the test file is as follows:

````
import 'react-native';
import React from 'react';
import Index from '../index.android.js';

// Note: test renderer must be required after react-native.
import renderer from 'react-test-renderer';
import mockCamera from '../__mocks__/react-native-camera'
jest.mock('react-native-camera', () => mockCamera);

it('renders correctly', () => {
  const tree = renderer.create(
    <Index />
  );
});
````

I've also added a project with proper commit history showing all the steps I've done and displaying the problem at had to: https://github.com/niom/jesttestapp

I've spent hours to figure out how I can get my tests working with react-native-camera but still can't get this to work.

samhunta, 2017-10-03:
This is probably not a react-native-camera issue. When using babel, jest automagically hoists jest.mock calls above import statements.

Remove the import for your mock and try - 
´´´
jest.mock('react-native-camera', () => require.requireActual('../__mocks__/react-native-camera').default
´´´.

niom, 2017-10-03:
Perfect! This solved the whole problem.

Thank you. Much appreciated.

I had a hunch that this was a stupid user error :)

closed 2017-10-03 by niom

samhunta, 2017-10-03:
Happy to help. This isn't a very well documented feature so hopefully its more apparent in the future :)

niom, 2017-10-03:
I Agree that it's not a very well documented feature. Now at least we have some way to help others facing the same problem by having this solved.

assafb81, 2017-10-15:
I encountered a problem when trying to work with the solution @samhunta:
My mock code is identical to what @niom wrote above.
It seems, from the error in the console, that the test does not goes throw the mock component but it goes to the component in the node_modules:
```
 ? Test suite failed to run

   TypeError: Cannot read property 'default' of undefined

     at Object.user;c:\**\App\node_modules\react-native-camera\index.js; (__tests__/index.ios.js:10:70)
     at Object.<anonymous> (src/components/componentA.js:5:24)
     at Object.<anonymous> (src/containers/componentB.js:11:20)
```

I tried to add the path `<rootDir>/__ mock _/react-native-camera.js` to jest configuration section in my package.json:
```
"jest": [
      .....
      "setupFiles": [
            "<rootDir>/test/globalFetch.js",
            "<rootDir>/__mock__/react-native-i18n.js",
            "<rootDir>/__mock__/react-native-push-notification.js"
      ]
]
```
But with no luck.

I am using:
```
        ....
        "react": "16.0.0-alpha.12",
        "react-native": "0.45.1",
        "react-native-camera": "^0.10.0",
        "jest": "20.0.4",
        ....
```

niom, 2017-10-17:
@assafb81 

My basic test file when using the mock is like this:

```
import 'react-native';
import React from 'react';
import Index from '../app/index'; //my awesome application that contains the react native camera

import renderer from 'react-test-renderer';
jest.mock('react-native-camera', () => require.requireActual('../__mocks__/react-native-camera').default)

it('renders correctly', () => {
  const tree = renderer.create(
    <Index />
  );
});
```

and the mock file it self is: 

```
import React from 'react';

const constants = constants =  {
  Aspect: {},
  BarCodeType: {},
  Type: {},
  CaptureMode: {},
  CaptureTarget: {},
  CaptureQuality: {},
  Orientation: {},
  FlashMode: {},
  TorchMode: {}
};

class Camera extends React.Component {

  static constants = constants
  render() {
    return null;
  }
}

Camera.constants = constants;

export default Camera;
```

assafb81, 2017-10-18:
@niom 
thanks for the reply.
1. I tried to import my mock camera using:
 `jest.mock('react-native-camera', () => require.requireActual('../__mocks__/react-native-camera').default)`
It did not solve my problem.
2. what is the purpose of this line in the mock file:
`const constants = constants =  {...}`
I removed one of `constans =` since I don't see why it is helpful.
3. what are the versions of JEST and RN are you using?

niom, 2017-10-18:
@assafb81 https://github.com/niom/jesttestapp/blob/master/package.json

assafb81, 2017-10-19:
@niom, when trying the test after updating my packages according to your example I've encounter new error:

```
FAIL  __tests__\index.ios.js
 ? Test suite failed to run

   TypeError: Cannot call a class as a function

     at _classCallCheck (__mocks__/react-native-camera.js:1:903)
     at Object.Camera [as default] (__mocks__/react-native-camera.js:15:88)
     at Object.user;C:\Projects\**\App\__mocks__\react-native-camera.js; (__tests__/index.ios.js:15:116)
     at Object.<anonymous> (src/components/componentA.js:5:24)
     at Object.<anonymous> (src/containers/ContainerA.js:11:20)
```