react-native-camera/react-native-camera #922: checkDeviceAuthorizationStatus is not work!!!

aspidvip, 2017-10-02:
Hello!!!

if(Camera.checkDeviceAuthorizationStatus) {
  console.log('yes');
} else {
  console.log('no');
}

PLease Help

SpidGiarz, 2017-10-03:
checkDeviceAuthorizationStatus is a promise method.

So your check would have to be something like:
```javascript
const someAsyncFunction = async () => {
 if(await Camera.checkDeviceAuthorizationStatus()) {
  console.log('yes');
 } else {
  console.log('no');
 }
}
```

mfahad1, 2017-10-03:
@SpidGiarz what about if on false we want to ask user to allow camera access , so if some other time user want to redo the permission .

SpidGiarz, 2017-10-03:
For Android leverage [PermissonsAndroid](https://facebook.github.io/react-native/docs/permissionsandroid.html) or a third party module to determine the camera permission state and act accordingly:
- if denied; request again with the appropriate rationale.
- if never_ask_again; explain the situation and, optionally, provide a button to quickly take them to your app's settings page.

For iOS if false then justify the need for the permission while providing a button to transport them to your app's settings page.

mfahad1, 2017-10-03:
@SpidGiarz thanks for a quick response , I am developing for IOS .. Is there really no way other then manually changing permission of the app ?

SpidGiarz, 2017-10-03:
Sadly none, other than providing them a compelling reason to manually grant you that coveted permission.

aspidvip, 2017-10-18:
componentDidMount(){
const someAsyncFunction = async () => {
 if(await Camera.checkDeviceAuthorizationStatus()) {
  console.log('yes');
 } else {
  console.log('no');
 }
}}
none work (((

jgcmarins, 2017-10-18:
@aspidvip check this: https://github.com/lwansbrough/react-native-camera/blob/master/README.md#requirements

aspidvip, 2017-10-19:
Camera.checkDeviceAuthorizationStatus()
    .then((result)=>{
      this.setState({
        accessFoto: result,
      });
})
this is work!!! thansk!!!

aspidvip, 2017-10-19:
please close topic! thansk)

closed 2017-10-22 by ptomasroos