react-native-camera/react-native-camera #925: how does the barcode scanning works ?

vivekebics, 2017-10-03:
I am not able to scan any barcode in android

sahir, 2017-10-03:
Please check this https://github.com/lwansbrough/react-native-camera/pull/874 you can also get complete solution for barcode-scanner https://github.com/sahir/react-native-barcode-reader

sahir, 2017-10-03:
`import React, { Component } from 'react';
import { View, AppRegistry, Text, Button } from 'react-native';
import Camera from 'react-native-camera';

export default class ScannerScene extends Component {
	constructor(props) {
		super(props);
		this.camera = null;
		this.barcodeCodes = [];

		this.state = {
			camera: {
				aspect: Camera.constants.Aspect.fill,
				captureTarget: Camera.constants.CaptureTarget.cameraRoll,
				type: Camera.constants.Type.back,
				orientation: Camera.constants.Orientation.auto,
				flashMode: Camera.constants.FlashMode.auto,
				barcodeFinderVisible: true
			}
		};
	}

	onBarCodeRead(scanResult) {
		console.warn(scanResult.type);
		console.warn(scanResult.data);
		if (scanResult.data != null) {
			if (!this.barcodeCodes.includes(scanResult.data)) {
				this.barcodeCodes.push(scanResult.data);
				console.warn('onBarCodeRead call');
			}
		}
		return;
	}

	render() {
		const styles = this.defaultStyles();
		return (
			<View style={styles.container}>
				<Camera
					ref={cam => {
						this.camera = cam;
					}}
					style={styles.preview}
					aspect={this.state.camera.aspect}
					captureTarget={this.state.camera.captureTarget}
					type={this.state.camera.type}
					flashMode={this.state.camera.flashMode}
					onFocusChanged={() => {}}
					onZoomChanged={() => {}}
					defaultTouchToFocus
					mirrorImage={false}
					barcodeFinderVisible={this.state.camera.barcodeFinderVisible}
					barcodeFinderWidth={280}
					barcodeFinderHeight={220}
					barcodeFinderBorderColor="red"
					barcodeFinderBorderWidth={2}
					onBarCodeRead={this.onBarCodeRead.bind(this)}
				/>
				<View style={[styles.overlay, styles.topOverlay]}>
					<Text style={styles.scanScreenMessage}>Please scan the barcode.</Text>
				</View>
				<View style={[styles.overlay, styles.bottomOverlay]}>
					<Button style={styles.enterBarcodeManualButton} title="Enter Barcode" />
				</View>
			</View>
		);
	}
	defaultStyles() {
		return {
			container: {
				flex: 1
			},
			preview: {
				flex: 1,
				justifyContent: 'flex-end',
				alignItems: 'center'
			},
			overlay: {
				position: 'absolute',
				padding: 16,
				right: 0,
				left: 0,
				alignItems: 'center'
			},
			topOverlay: {
				top: 0,
				flex: 1,
				flexDirection: 'row',
				justifyContent: 'space-between',
				alignItems: 'center'
			},
			bottomOverlay: {
				bottom: 0,
				backgroundColor: 'rgba(0,0,0,0.4)',
				flexDirection: 'row',
				justifyContent: 'center',
				alignItems: 'center'
			},
			enterBarcodeManualButton: {
				padding: 15,
				backgroundColor: 'white',
				borderRadius: 40
			},
			scanScreenMessage: {
				fontSize: 14,
				color: 'white',
				textAlign: 'center',
				alignItems: 'center',
				justifyContent: 'center'
			}
		};
	}
}

AppRegistry.registerComponent('barcode-reader', () => ScannerScene);`

vivekebics, 2017-10-03:
Thanks sahir.

ashishrokr, 2017-10-10:
how do i play default sound on Barcode/QrCode read ?

simistern, 2019-09-25:
You can hook the sound onto the barcode reading listener.

[Barcode Example here](https://github.com/react-native-community/react-native-camera/blob/master/examples/mlkit/App.js)

closed 2019-09-25 by simistern