react-native-camera/react-native-camera #926: Scanning QR code crashes app

justynspooner, 2017-10-03:
This QR code causes a crash.

[image]

in `RCTCameraManager.m`

in the method `- (void)captureOutput:(AVCaptureOutput *)captureOutput didOutputMetadataObjects:(NSArray *)metadataObjects fromConnection:(AVCaptureConnection *)connection`

```
@"data": metadata.stringValue,
```

calling stringValue causes the crash.

fokoz, 2017-10-22:
Mine crashes only when specify `barCodeTypes` to qr but if not specify, it is fine.

prajnarobo, 2018-08-14:
Any solution for the above issue? App is crashing even after removing barCodeTypes

hariDasu, 2019-02-07:
this works with the above merged PR, this issue can be closed now @prajnarobo @justynspooner

sibelius, 2019-02-07:
use RNCamera instead of RCTCamera

closed 2019-02-07 by sibelius