react-native-camera/react-native-camera #930: Barcode scan not work on android in landscape orientation

volodyad, 2017-10-04:
Barcode scan only works in portrait orientation on android, is this by desing?

alexpanov, 2017-10-10:
PR #448 is sitting there for almost a year now. Wow.

niom, 2017-11-09:
There's even a new attempt to fix this through PR https://github.com/lwansbrough/react-native-camera/pull/949. 

Our project would also love to have Barcode scanning working with landscape orientation. 

Any news to get this feature done / fixed?

Dror-Bar, 2018-02-08:
Still not working for me in ^0.12.0

platonish, 2018-06-30:
Any updates on this? Still doesn't work on Android (`react-native-camera v1.1.4`)

Dror-Bar, 2018-07-01:
@platonish 
For me, RNCamera from version 1.0.3 works for me in Android to scan for barcode.

I believe, but not sure, that there's a different bug in this version in Android where images taken in landscape mode will be rotated 90 degrees. But it might not be relevant to you if you only need barcode scan.

cooperjbrandon, 2018-07-10:
I was seeing this problem as well. Not sure if this is helpful to you, but for me it was because I declared my device to be locked in Portrait mode.

[image]

The `rotation` parameter was always passed in as 0: https://github.com/react-native-community/react-native-camera/blob/v1.1.4/android/src/main/java/org/reactnative/camera/RNCameraView.java#L122 , regardless of what orientation the phone was actually in.

So `correctRotation` ended up always incorrectly being 90 https://github.com/react-native-community/react-native-camera/blob/v1.1.4/android/src/main/java/org/reactnative/camera/RNCameraView.java#L123

So it would always try to rotate the image https://github.com/react-native-community/react-native-camera/blob/v1.1.4/android/src/main/java/org/reactnative/camera/RNCameraView.java#L127 .

However, looks like this PR https://github.com/react-native-community/react-native-camera/pull/1667/files takes care of the issue, which is in the newest 1.1.5 release.

closed 2018-09-23 by SimonErm

Dror-Bar, 2018-09-25:
This doesn't really help if you want your app to be locked in Portrait mode, doesn't it?
Hopefully it will be fixed soon, however.

Dror-Bar, 2018-09-25:
hi SimonErm - 
My apologize, I haven't checked the latest PR when writing my comment, I'll see if I can test it soon but I do take your word that it indeed fixes the problem, looking forward to the newest release.

subh9932865811, 2020-02-20:
How can i fixed problem when mobile roration on scanner not working

eamonn-alphin, 2021-02-17:
was this ever fixed or am I doing something wrong because it's still not scanning in landscape mode...

fellenabmb, 2021-04-09:
If you're here wondering if this works, as of now, it still doesn't if you have your app locked in portrait mode.
It'd be really nice to see a fix for this!.

fellenabmb, 2021-04-09:
If you're here wondering if this works, as of now, it still doesn't if you have your app locked in portrait mode.
It'd be really nice to be able to make it work in landscape mode, regardless of the mode that the app is locked in 😄 !.

cdenadai, 2021-05-11:
Why this is closed if don't work?