react-native-camera/react-native-camera #933: keep getting cannot find symbol when compiling java.

1mike12, 2017-10-05:
react native 0.49.1
windows 10 64

tried doing a gradle clean without luck. 

also fyi my main java extends wix's NavigationApplication, and I don't really need the mReactNativeHost but too lazy to keep deleting after each react-native link. 
```
 error: cannot find symbol
             new RCTCameraPackage(),
                 ^
  symbol:   class RCTCameraPackage
  location: class MainApplication
3 errors
:app:compileDebugJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details
.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --d
ebug option to get more log output.

BUILD FAILED

```

MainApplication.java
```java
package com.phoneafriend;

import android.app.Application;

import com.facebook.react.ReactApplication;
import com.lwansbrough.RCTCamera.RCTCameraPackage;
import com.oney.WebRTCModule.WebRTCModulePackage;
import com.oblador.vectoricons.VectorIconsPackage;
import com.lugg.ReactNativeConfig.ReactNativeConfigPackage;
import com.oblador.vectoricons.VectorIconsPackage;
import com.lugg.ReactNativeConfig.ReactNativeConfigPackage;
import com.facebook.react.ReactNativeHost;
import com.facebook.react.ReactPackage;
import com.facebook.react.shell.MainReactPackage;
import com.facebook.soloader.SoLoader;
import com.oney.WebRTCModule.WebRTCModulePackage;
import com.reactnativenavigation.NavigationApplication;

import java.util.Arrays;
import java.util.List;

public class MainApplication extends NavigationApplication {

    private final ReactNativeHost mReactNativeHost = new ReactNativeHost(this) {
        @Override
        public boolean getUseDeveloperSupport() {
            return BuildConfig.DEBUG;
        }

        @Override
        protected List < ReactPackage > getPackages() {
            return Arrays. < ReactPackage > asList(
                new MainReactPackage(),
                new RCTCameraPackage(),
                new VectorIconsPackage(),
                new ReactNativeConfigPackage(),
                new WebRTCModulePackage()
            );
        }
    };

    @Override
    public boolean isDebug() {
        // Make sure you are using BuildConfig from your own application
        return BuildConfig.DEBUG;
    }

    protected List < ReactPackage > getPackages() {
        // Add additional packages you require here
        // No need to add RnnPackage and MainReactPackage
        return Arrays. < ReactPackage > asList(
            new RCTCameraPackage(),
            new VectorIconsPackage(),
            new ReactNativeConfigPackage(),
            new WebRTCModulePackage()
        );
    }

    @Override
    public List < ReactPackage > createAdditionalReactPackages() {
        return getPackages();
    }

    @Override
    public ReactNativeHost getReactNativeHost() {
        return mReactNativeHost;
    }

    @Override
    public void onCreate() {
        super.onCreate();
        SoLoader.init(this, /* native exopackage */ false);
    }
}
```

settings.gradle
```gradle
rootProject.name = 'example'
include ':react-native-webrtc'
project(':react-native-webrtc').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-webrtc/android')
include ':react-native-vector-icons'
project(':react-native-vector-icons').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-vector-icons/android')
include ':react-native-navigation'
project(':react-native-navigation').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-navigation/android/app')
include ':react-native-config'
project(':react-native-config').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-config/android')
include ':react-native-vector-icons'
project(':react-native-vector-icons').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-vector-icons/android')
include ':react-native-config'
project(':react-native-config').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-config/android')

include ':app'
include ':WebRTCModule', ':app'
project(':WebRTCModule').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-webrtc/android')

 include ':react-native-navigation'
 project(':react-native-navigation').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-navigation/android/app/')
```

pongponglau, 2017-10-22:
I encounter the same problem when building the Android app with RN 0.49.

All third party packages in MainApplication.java are "Symbol not found".

May i ask if you have any solution for your problem?

Thank you

1mike12, 2017-10-23:
I got it working, but I don't even remember how. I think I just rm rf'd node_modules and restarted the install and build.

igorbrandao, 2017-11-07:
Just bumped into the same problem... Any updates?

igorbrandao, 2017-11-07:
@1mike12 and @pongponglau I have found the problem! By following the manual link instructions for android, I saw that `react-native link` does all the steps except the 4th one, that adds `compile project(':react-native-camera')` to the file `android/app/build.gradle`.

Doing that solved this problem for me, hope it does for you as well. 😄

atljoseph, 2018-02-23:
My solution here: #228

sibelius, 2018-02-23:
Fixed on master

closed 2018-02-23 by sibelius