react-native-camera/react-native-camera #934: Fresh install broke my app

ODelibalta, 2017-10-05:
I ran the npm install command then the link command. Now I get this error 

```
2017-10-04 23:37:16.663 xcodebuild[35725:9409054] CFPropertyListCreateFromXMLData(): Old-style plist parser: missing semicolon in dictionary on line 1059. Parsing will be abandoned. Break on _CFPropertyListMissingSemicolon to debug.

2017-10-04 23:37:16.667 xcodebuild[35725:9409054] CFPropertyListCreateFromXMLData(): Old-style plist parser: missing semicolon in dictionary on line 1059. Parsing will be abandoned. Break on _CFPropertyListMissingSemicolon to debug.

2017-10-04 23:37:16.668 xcodebuild[35725:9409054] Error Domain=NSCocoaErrorDomain Code=3840 "Unexpected character / at line 1" UserInfo={NSDebugDescription=Unexpected character / at line 1, kCFPropertyListOldStyleParsingError=Error Domain=NSCocoaErrorDomain Code=3840 "Missing ';' on line 1059" UserInfo={NSDebugDescription=Missing ';' on line 1059}}
```

I am not sure what to do at this point and I'd greatly appreciate any help.

jgcmarins, 2017-10-05:
Hello @ODelibalta could you provide the link command? Thanks :)

jgcmarins, 2017-10-05:
hmmm, seems ok.
Did you try to rebuild the app in xcode?
First, clean the build, then build the app from xcode. Make sure to reset the cache. To do that, add this line to your `package.json` scripts:
```
"clear": "node node_modules/react-native/local-cli/cli.js start --reset-cache"
```

ODelibalta, 2017-10-05:
I've tried to open the project in xcode but xcode could not open the project. 

```
xcodebuild: error: Unable to read project 'vivaa.xcodeproj'.
	Reason: Project /Users/Impulsive/WebSites/flink/natives/mobapp/ios/mobapp.xcodeproj cannot be opened because the project file cannot be parsed.
```
I am not experienced in rnative but if I were to guess, I think linking caused a syntax error in my pod files? Just guessing lol I do not know how I'd fix that though :)

jgcmarins, 2017-10-05:
I guess your problem isn't related to react-native-camera.
Let's keep this issue opened for a few days, but I think we can close it.

ODelibalta, 2017-10-05:
@jgcmarins What are your thoughts as to what this might be related to? After an hour of development on the app in functional state, I've ran the two commands for the install and now it will not open.

ODelibalta, 2017-10-06:
I replaced my files from a backup without the camera lib then ran npm install which my backup did not have. 
Everything worked. 
Now installed the lib through npm. Manually linked everything. iOS works. On simulator it opens and I see the capture button just no camera as expected given its simulator. 
On Android simulator, it compiles, when it tries to open the camera page, it says `cannot read Aspect of undefined` and I think it is the line `aspect={Camera.constants.Aspect.fill}>
` from the example code.  Is it because it is a emulator and it does not have camera? I have not tried on an actual android phone yet.

xavs, 2017-10-10:
I'm doing a brand new app with 
`react-native init myapp`
then i add the react-native-camera with the automatic instructions ( I then check all the things in the  manual install are actually in place )
Then I get an error like @ODelibalta and I have figured that 
`import { Camera } from 'react-native-camera'` is not working, and actually a `var Camera = require('react-native-camera')` works.

Mate38, 2017-11-03:
My app was running perfectly, it was just running the commands npm install react-native-camera --save and react-native link react-native-camera which started to display the error below:

[image]

I created an app from scratch just to test, ran perfect, after running the commands I received the same error.

jgcmarins, 2017-11-03:
Hello @Mate38 , react-native link can sometimes not link well.
Did you followed README instructions?
Did you checked Android files to see if they were linked properly?

Mate38, 2017-11-03:
@jgcmarins I checked and the link created everything exactly as described for the manual process, except the permissions, but even adding these manually continues all the same.

Mate38, 2017-11-03:
One fact that I found strange during the installation was this:

```
D:\APPs\Camera>react-native run-android
Scanning folders for symlinks in D:\APPs\Camera\node_modules (18ms)
Starting JS server...
Building and installing the app on the device (cd android && gradlew.bat install
Debug)...
Incremental java compilation is an incubating feature.
:app:preBuild UP-TO-DATE
:app:preDebugBuild UP-TO-DATE
:app:checkDebugManifest
:app:preReleaseBuild UP-TO-DATE
:app:prepareComAndroidSupportAppcompatV72301Library UP-TO-DATE
:app:prepareComAndroidSupportSupportV42301Library UP-TO-DATE
:app:prepareComFacebookFbuiTextlayoutbuilderTextlayoutbuilder100Library UP-TO-DA
TE
:app:prepareComFacebookFrescoDrawee130Library UP-TO-DATE
:app:prepareComFacebookFrescoFbcore130Library UP-TO-DATE
:app:prepareComFacebookFrescoFresco130Library UP-TO-DATE
:app:prepareComFacebookFrescoImagepipeline130Library UP-TO-DATE
:app:prepareComFacebookFrescoImagepipelineBase130Library UP-TO-DATE
:app:prepareComFacebookFrescoImagepipelineOkhttp3130Library UP-TO-DATE
:app:prepareComFacebookReactReactNative0500Library UP-TO-DATE
:app:prepareComFacebookSoloaderSoloader010Library UP-TO-DATE
:app:prepareOrgWebkitAndroidJscR174650Library UP-TO-DATE
:app:prepareDebugDependencies
:app:compileDebugAidl UP-TO-DATE
:app:compileDebugRenderscript UP-TO-DATE
:app:generateDebugBuildConfig UP-TO-DATE
:app:mergeDebugShaders
:app:compileDebugShaders
:app:generateDebugAssets
:app:mergeDebugAssets
:app:generateDebugResValues UP-TO-DATE
:app:generateDebugResources UP-TO-DATE
:app:mergeDebugResources UP-TO-DATE
:app:bundleDebugJsAndAssets SKIPPED
:app:processDebugManifest UP-TO-DATE
:app:processDebugResources UP-TO-DATE
:app:generateDebugSources UP-TO-DATE
:app:incrementalDebugJavaCompilationSafeguard
:app:compileDebugJavaWithJavac
:app:compileDebugJavaWithJavac - is not incremental (e.g. outputs have changed,
no previous execution, etc.).
:app:compileDebugNdk UP-TO-DATE
:app:compileDebugSources
:app:transformClassesWithDexForDebug
Running dex in-process requires build tools 23.0.2.
For faster builds update this project to use the latest build tools.
:app:mergeDebugJniLibFolders
:app:transformNative_libsWithMergeJniLibsForDebug
:app:transformNative_libsWithStripDebugSymbolForDebug
:app:processDebugJavaRes UP-TO-DATE
:app:transformResourcesWithMergeJavaResForDebug
:app:validateSigningDebug
:app:packageDebug
:app:assembleDebug
:app:installDebug
Installing APK 'app-debug.apk' on 'Nexus_5X_API_26(AVD) - 8.0.0' for app:debug
Installed on 1 device.

BUILD SUCCESSFUL

Total time: 49.035 secs
Running adb -s emulator-5554 reverse tcp:8081 tcp:8081
Starting the app on emulator-5554 (adb -s emulator-5554 shell am start -n com.ca
mera/com.camera.MainActivity)...
Starting: Intent { cmp=com.camera/.MainActivity }
^CDeseja finalizar o arquivo em lotes (S/N)? s

D:\APPs\Camera>npm install react-native-camera --save
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\jest.cmd as it w
asn't installed by D:\APPs\Camera\node_modules\jest
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\jest as it wasn'
t installed by D:\APPs\Camera\node_modules\jest
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\json5.cmd as it
wasn't installed by D:\APPs\Camera\node_modules\json5
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\json5 as it wasn
't installed by D:\APPs\Camera\node_modules\json5
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\esvalidate.cmd a
s it wasn't installed by D:\APPs\Camera\node_modules\esprima
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\esparse.cmd as i
t wasn't installed by D:\APPs\Camera\node_modules\esprima
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\esvalidate as it
 wasn't installed by D:\APPs\Camera\node_modules\esprima
npm WARN gentlyRm not removing D:\APPs\Camera\node_modules\.bin\esparse as it wa
sn't installed by D:\APPs\Camera\node_modules\esprima
npm notice created a lockfile as package-lock.json. You should commit this file.

+ react-native-camera@0.10.0
added 1 package, removed 674 packages and updated 16 packages in 14.076s

D:\APPs\Camera>react-native run-android
Command `run-android` unrecognized. Make sure that you have run `npm install` an
d that you are inside a react-native project.

D:\APPs\Camera>
```

After running `npm-install` this one presented the same error that I described earlier.

sibelius, 2018-01-07:
this is not related to this package

try to use NVM to make sure your node environment is good

closed 2018-01-07 by sibelius