react-native-camera/react-native-camera #936: Broke with react 16.0.0.  @ React.PropTypes.oneOfType

skathayat, 2017-10-05:
React v. 16.0.0 comes up with separate 'prop-types module. Having said that it is not included in React itself. That means when react-native-camera is used with react v16.0.0, following error appears

"Cannot read property oneOfType" of undefined.

Any fix for this?

I think the solution will be importing PropType from 'prop-types' instead of 'react

Currently in react-native-camera - import React, { Component, PropTypes } from 'react';
Should be - import PropTypes from 'prop-types'

jgcmarins, 2017-10-05:
I guess that it's better to wait for React Native updates to React 16, then you try to update yours project React version. For now, keep React in the same version of React Native's React version.

djalmaaraujo, 2017-10-05:
That would be amazing.

Getting errors because of the proptypes too.

[image]

djalmaaraujo, 2017-10-05:
Looks like you already did on master... @lwansbrough ?

skathayat, 2017-10-06:
v0.10.0 seems to be working with react 16!

djalmaaraujo, 2017-10-06:
Working for me. I am using master.

joshkuhar, 2017-10-09:
I updated the `dependencies` in my package.json so that it points to the github repo. Also, maybe adding 'prop-types' to both `dependencies` and devDependencies`. However, I got it working by just pointing to the repo, like below. 

```	"dependencies": {
		"react": "16.0.0-beta.5",
		"react-native": "0.49.1",
		"react-native-camera": "https://github.com/lwansbrough/react-native-camera.git"
	},
```

darioodiaz, 2017-11-15:
I have the same problem but because I follow the installation steps using version 0.6, I unlink and remove that from my project, then install react-native-camera with no version specific, link again and works for me. I suggest remove the step "for react-native >= 0.4 use react-native-camera@0.6" or if you prefer I can modify it and make the PR

sibelius, 2017-12-09:
working fine on master, and latest version of RN

closed 2017-12-09 by sibelius

lakhman, 2017-12-14:
I had the same issue as @darioodiaz, Uninstall and reinstall without any version.

``` bash
# uninstall
react-native unlink react-native-camera
npm uninstall react-native-camera --save

# reinstall with no version
npm install react-native-camera --save
react-native link react-native-camera
```

example `package.json`.

``` json
"prop-types": "^15.6.0",
"react": "16.0.0",
"react-native": "0.51.0",
"react-native-camera": "^0.12.0",
```

Everything works as expected now.
Hope this helps!

hungdev, 2018-02-10:
@lakhman  thank you. it worked for me.