react-native-camera/react-native-camera #939: Location isnt added to data on IOS 11.01

AnnyA95, 2017-10-09:
I want to add the location of the photo so but it isnt added to the data object
[image]
[image]

taiki-t, 2017-10-28:
Yeah, It's happening to me as well. Still looking for a solution.

koswarabilly, 2017-11-17:
Android also got the same issue
in my project the app crash immediately when trying to capture because of metadata

wbyoung, 2018-04-05:
It looks like you need to set `location` on the `metadata` rather than on options directly. I just opened #1437 to try to make that more clear.

closed 2018-04-05 by sibelius

taiki-t, 2018-04-06:
> it looks like you need to set location on the metadata rather than on options directly.

That's true. I wasn't looking at the code in the issue description well. But the problem still remains. Even though you set `location` in `metadata`, in iOS 11.x (now 11.2.6) doesn't save photo with location.