react-native-camera/react-native-camera #940: java.lang.RuntimeException: setParameters failed

sibelius, 2017-10-09:
android 7

```jsx
Fatal Exception: java.lang.RuntimeException: setParameters failed
       at android.hardware.Camera.native_setParameters(Camera.java)
       at android.hardware.Camera.setParameters(Camera.java:2021)
       at com.lwansbrough.RCTCamera.RCTCamera.setCaptureQuality(RCTCamera.java:246)
       at com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation(RCTCameraModule.java:516)
       at com.lwansbrough.RCTCamera.RCTCameraModule.access$100(RCTCameraModule.java:38)
       at com.lwansbrough.RCTCamera.RCTCameraModule$2.orientationEvent(RCTCameraModule.java:496)
       at com.lwansbrough.RCTCamera.RCTSensorOrientationChecker$Listener.onSensorChanged(RCTSensorOrientationChecker.java:64)
       at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:701)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:323)
       at android.os.Looper.loop(Looper.java:139)
       at android.app.ActivityThread.main(ActivityThread.java:6097)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
```

brunsy, 2018-03-19:
I get this on some devices when not specifying any settings for recording video. 
Muting or explicitly setting the recording quality for recordAsync seemed to bypass this issue for me.

carloscuesta, 2018-09-21:
Got the same error too!

JslinSir, 2018-10-22:
+1

 android.hardware.Camera.native_setParameters(Native Method)
--
 android.hardware.Camera.setParameters(Camera.java:1940)
 com.google.android.cameraview.Camera1.setDisplayOrientation(Camera1.java:454)
 com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:121)
 com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged(DisplayOrientationDetector.java:86)
 com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:62)
android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:845)
android.os.MessageQueue.nativePollOnce(Native Method)
android.os.MessageQueue.next(MessageQueue.java:323)
 android.os.Looper.loop(Looper.java:136)
 android.app.ActivityThread.main(ActivityThread.java:6776)
 java.lang.reflect.Method.invoke(Native Method)
 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1518)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)

[image]

[image]

augustl, 2018-11-30:
We just got one as well. Android 8.1.0, device SM-G390F, React Native 0.57.4, react-native-camera v1.4.1. Sentry also reported "orientation: landscape".

```
java.lang.RuntimeException: setParameters failed
    at android.hardware.Camera.native_setParameters(Camera.java)
    at android.hardware.Camera.setParameters(Camera.java:1945)
    at com.google.android.cameraview.Camera1.setDisplayOrientation(Camera1.java:458)
    at com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:123)
    at com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged(DisplayOrientationDetector.java:86)
    at com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:62)
    at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
    at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:964)
    at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
    at android.os.MessageQueue.next(MessageQueue.java:325)
    at android.os.Looper.loop(Looper.java:142)
    at android.app.ActivityThread.main(ActivityThread.java:7000)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)
```

sraka1, 2019-02-12:
We're also receiving this in Fabric, but are unable to replicate...has anyone seen this problem?

carloscuesta, 2019-02-12:
@sraka1 In my particular case it happened because a user denied the camera permissions and then he tried to enter to the screen that showed up the Camera

joshuakelly, 2019-02-25:
Any news on this? Have been receiving a similar error.

GerardCasadevall, 2019-02-28:
> We're also receiving this in Fabric, but are unable to replicate...has anyone seen this problem?

Idem!!!! Any news on this?

sraka1, 2019-03-05:
@n1ru4l Did you happen to dig around this issue any further?

sibelius, 2019-03-12:
anybody can help with this one?

joshuakelly, 2019-03-12:
has anyone picked up this work or is it just sitting here?

sibelius, 2019-03-12:
just sitting here

drewg233, 2019-03-12:
I ran across this issue as well this morning. I am no android developer, but this seemed to fixed it for me.

Try adding the following on your `AndroidManifest.xml` right under all of your uses-permissions statements:

```
    <uses-feature android:name="android.hardware.camera.setParameters" android:required="false" />
    <uses-feature android:name="android.hardware.camera.autofocus" android:required="false" />
```

sibelius, 2019-03-12:
@drewg233 can you add this to docs or FAQ?

joshuakelly, 2019-03-21:
> I ran across this issue as well this morning. I am no android developer, but this seemed to fixed it for me.
>
> Try adding the following on your `AndroidManifest.xml` right under all of your [...]

This didn't work for me unfortunately. Still having users with that error.

sraka1, 2019-03-22:
Same here. Did not help, crash is still occurring.

benhartouz, 2019-03-29:
Same error here

zaguiini, 2019-03-29:
I think it's not directly related to the RN Camera Javascript module? Is an Android/Java error.

Googling, I've found this SO answer: https://stackoverflow.com/questions/33579918/camera-java-lang-runtimeexception-setparameters-failed

It _looks_ that, perhaps, the accepted answer might be a good one to try...

sibelius, 2019-04-01:
we need to implement this

```jsx
List<Size> allSizes = param.getSupportedPictureSizes();
Camera.Size size = allSizes.get(0); // get top size
for (int i = 0; i < allSizes.size(); i++) {
     if (allSizes.get(i).width > size.width)
       size = allSizes.get(i);
 }
//set max Picture Size
 params.setPictureSize(size.width, size.height);
```

sibelius, 2019-04-01:
the method is here

https://github.com/react-native-community/react-native-camera/issues/1535#issuecomment-478408696

you can get this using this method

https://github.com/react-native-community/react-native-camera/blob/ca35ac609d2e1d0b93f06056f3d1a78424e25dca/src/RNCamera.js#L400

`getAvailablePictureSizes`

check this example code

```jsx
onCameraReady = async () => {
    if (this.camera) {
      const ratios = await this.camera.getSupportedRatiosAsync();

      const sizes = await this.camera.getAvailablePictureSizes();

      console.log('onCameraReady ratios: ', ratios);
      console.log('onCameraReady sizes: ', sizes);
    }
  };
```

result:

```
'onCameraReady ratios: ', [ '3:2', '4:3', '5:3', '11:9', '16:9' ]
'onCameraReady sizes: ', [ '1280x720', '1920x1080' ]
```

it will only return sizes based on the current ratio 

can somebody validate if using the correct available size solves the issue?

sraka1, 2019-04-01:
@sibelius me and a couple users are getting the setParameters failed error on the `setDisplayOrientation` call. Settings the picture size most likely won't affect this (although I am not sure).

emilemoureau, 2019-04-15:
Got the same error too! RN 0.59.2 & Android 9

sibelius, 2019-04-15:
@sraka1 I think the problem is when `setDisplayOrientation` happens, it will try to set a parameter that is invalid

we need to change the camera ratio or size to fix this, or to be safe

can you work on a PR? and then we can review

sibelius, 2019-04-18:
RCTCamera is removed https://github.com/react-native-community/react-native-camera/issues/1890

Migrate to latest version of RNCamera

sraka1, 2019-04-18:
@sibelius this occurs on RNCamera and most reports are using RNCamera, please reopen.

sraka1, 2019-04-18:
I managed to replicate this somehow.

Here's what logcat says
```
CameraClient: attempt to use a locked camera from a different process (old pid 0, new pid 26537)
AndroidRuntime: Shutting down VM
04-18 11:29:54.941 26537 26537 D CrashlyticsCore: Crashlytics is handling uncaught exception "java.lang.RuntimeException: setParameters failed" from thread main
```

Looking at the [Android source here](https://android.googlesource.com/platform/frameworks/av/+/ffbc80f/services/camera/libcameraservice/api1/CameraClient.cpp), it seems that pid 0 means that camera is unlocked, so I'm not sure why checkPid() is being called. In other words, the camera is being unlocked and then setParameters called on it (which seems to be disallowed). They could've fixed this in the Android source by automatically calling lock() if the camera is unlocked, however they don't. It seems it's the clients responsibility to maintain the lock.

Investigating further soon.

agrass, 2019-05-28:
Hi @sraka1, 

How do you replicate this error?  I'm trying to replicate this in my phone but was not possible, but in a Samsung Galaxy Tab A 7.0 is happening all the time when I scan a barcode and then in a few seconds when I close the camera the app crash with this error. Starts when I updated to react-native-camera 2.3.0 and I added autoFocusPoint, but not sure if is for that or not.

Linoa65, 2019-06-12:
Same here ! It happens when I change device orientation, then the app crashes with this error : 

```

Fatal Exception: java.lang.RuntimeException 
setParameters failed
--
  | android.hardware.Camera.native_setParameters (Camera.java)
  | com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged (DisplayOrientationDetector.java:90)
  | android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged (OrientationEventListener.java:157)
  | android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent (SystemSensorManager.java:851)
  | com.android.internal.os.ZygoteInit.main (ZygoteInit.java:921)

```

agrass, 2019-06-12:
What version of react-native-camera are you using? Now should be fixed with last release, you should check with ^2.10.0

agrass, 2019-06-12:
With this PR #2288

virgule-xyz, 2019-06-30:
Still there, especially with QR Code, and also fires some kind of texture errors... I'm still digging this bugs, which seems to be a JAVA related bug...

MateusAndrade, 2019-07-01:
@virgule-xyz please let us know if you find anything.

virgule-xyz, 2019-07-01:
So here the story. I've got one scanner in a screen, followed by another, then a dashboard screen. All in a navigation stack. From here, if I switch to another app, like Maps, then come back to the app, it crashes. point. I have finally used a top context to handle and manage a 'show scanner' state for the two scanners so only one is able to be alive during the RN app lifecycle (I use only hooks for this app) and there is no more crashes. I now also put those two screens in switch navigation.
I must rise that under iOS there is no such problems.
There are some issues with the Orientation AND with the setPreviewTexture in the Java code.
Hard to tell more because it is a side project where I need to make relevant progress...

patelarunyo11, 2019-08-23:
By setting the state of the Image in the function,I get rid out of this problem/bug.You can use setState for this bug.

TechNov, 2019-10-05:
this solve my issue
==============================
Issue is caused by:
params.setPictureSize(1200, 900);
because required size is not suppoerted by Camera.

Use getSupportedPictureSizes to get all available preview sizes.

To check which is maximum picture size available from camera:

List<Size> allSizes = param.getSupportedPictureSizes();
Camera.Size size = allSizes.get(0); // get top size
for (int i = 0; i < allSizes.size(); i++) {
     if (allSizes.get(i).width > size.width)
       size = allSizes.get(i);
 }
//set max Picture Size
 params.setPictureSize(size.width, size.height);
source [Here](https://stackoverflow.com/questions/33579918/camera-java-lang-runtimeexception-setparameters-failed)

closed 2019-12-11 by stale[bot]