react-native-camera/react-native-camera #941: Capture image with specific height, with and borderRadius?

bvv218, 2017-10-10:
Is there a way to set up width and height for an image to be captured? For example, if I need a picture with width: 100 and height: 100, and borderRadius of 50, is there a way to capture and save it as a rounded image?

josenaves, 2017-10-10:
Take a look at this https://github.com/lwansbrough/react-native-camera#capturequality

bvv218, 2017-10-12:
@josenaves It only allows to change picture quality, not width/height/shape of the image to make it rounded.

nakulkundaliya, 2017-11-16:
@bvv218  Did you find any solutions of it?

allaire, 2018-11-07:
For border radius, you must add RNCamera to a view, and use `overflow: hidden`, exemple:

```
<View style={styles.cameraView}>
  <RNCamera
    ref={(ref) => {
      this.camera = ref;
    }}
    style={styles.camera}
    type={RNCamera.Constants.Type.front}
    permissionDialogTitle="Permission to use camera"
    permissionDialogMessage="We need your permission to use your camera phone"
  />
</View>
```

```
...
  cameraView: {
    alignSelf: 'center',
    width: scale(100),
    height: scale(100),
    marginBottom: scale(20),
    borderRadius: scale(100) / 2,
    backgroundColor: colors.grayLight,
    overflow: 'hidden',
  },
  camera: {
    flex: 1,
  },
...
```

closed 2019-06-08 by stale[bot]