react-native-camera/react-native-camera #942: Saving picture take 3-4 seconds

Borisboky, 2017-10-10:
The versions that I'm using are:
```
"react-native-camera": "^0.10.0",
"react-native": "0.48.3"
```

The saving images is **very very** slow. In my app the user can take up to 15 photos in a row, but with this issue it is almost not possible. They need to much time to do it.

When I take photo I hear sound and then it takes a 3-4 seconds to save it.

Is there any solution to this?

sibelius, 2018-02-05:
are u using RNCamera or RCTCamera?

RNCamera is very fast when just creating a uri file

Darwex, 2018-02-06:
Hey.

I was using the old Camera but now I switched to RNCamera but it seems that
it does not have constants.

import {RNCamera} from 'react-native-camera'

...

RNCamera.constants.Aspect.fill

Does not seem to be working.

Thanks a lot.
David.

sibelius, 2018-02-06:
Check migration guide

Darwex, 2018-02-06:
Hmm it runs now but hangs on capture const cameraData = await this.camera.takePictureAsync(). 
I will try to play around with it.

Darwex, 2018-02-06:
`
handleCapturePhoto = async() => {
    try {
      await this.setState({isCapturing: true})

      const cameraData = await this.camera.takePictureAsync()

      this.setState({
        takenPhoto: cameraData.path,
        isCapturing: false
      })
    } catch (e) {
      console.log(e)
    }
  }
`

`
     <RNCamera
        ref={this.handleCreateCameraRef}
        style={styles.camera}
        type={this.state.cameraType}
      >
        {this.renderActivityIndicator()}
        {goBackButton}
        {capturePhotoButton}
        {switchCameraButton}
      </RNCamera>
`

Error: takePictureAsync: Expected a Camera component

ospfranco, 2018-04-03:
added skipProcessing flag (android only) for the takePictureAsync call params, this should make picture snapping super fast, but might break rotation on some of your apps.

Please give it a go on the master branch.

lucasluca, 2020-02-06:
i was using this config:
const options = { quality: 0.1, base64: false, forceUpOrientation: true, fixOrientation: true, skipProcessing: true };

and then i removed {forceUpOrientation: true, fixOrientation: true}, from this object and got it working really fast

DevelopSmith, 2020-07-29:
For me, it gets stuck when it reaches this line (on iPhone via TestFlight):
`const data = await camera.current.takePictureAsync(options)`

closed 2020-10-04 by stale[bot]

devartwa, 2020-12-04:
i removed fixOrientation: false and worked for me.