react-native-camera/react-native-camera #945: Options to select preview size.

asdacap, 2017-10-13:
Hi there. So I was figuring out why the QR scanning performance is very poor on Android, and I found out that `ReaderAsyncTask` took nearly 2.5 second on my phone (which is a Xiaomi Mi3 with Snapdragon 800). Specifying the `barCodeTypes` to `['qr']` reduce the time to 1.5 second, but that was still pretty bad. 

It turns out, the preview size was 3840 by 2160 pixel. Which is unnecessarily large, especially considering that this device only have a 1080p screen. Efficiency aside, because the `MultiFormatReader` uses the preview data, it takes a lot of time because it have a lot of data to process which make the latency very poor.

Setting the maximum preview size to 720p at https://github.com/lwansbrough/react-native-camera/blob/4b4160801e9d52699d0d40c6d9614bdc3754a284/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java#L401 reduces the processing time to 0.15 second which is acceptable not to mention a whole lot faster than before.

So it would be nice if there is an option to specify the preview size for some use case like QR scanning which don't really need a high quality preview, but can benefit from the reduced latency. As an added bonus, it can help reduce power consumption. For now, I'll just manually patch `node_modules`.

CodingPapi, 2018-07-02:
config for preview size is necessary

laurentrivard, 2018-10-11:
Interesting in this as well. The camera takes a while to initialize, and it also slows down every frame!

closed 2019-04-18 by sibelius