react-native-camera/react-native-camera #947: Realm JS 'write' is synchronously?

kosiakMD, 2017-10-17:
Is Realm JS 'write' method is synchronously?
'Cause I can't understand those things:
When I 'read' Realm DB and then handle previously read information it goes synchronously and next code is dunned step by step.
But when I put 'write' some data and run next some code (redux dispatch to update Store Data) it's not up to date.
Only Notifications via 'addListener' works, but those updates are often slowly even in NOT Debugger Mode.
So if 'write' is handled asynchronously, how to get simple callback directly after it without 'listeners'?
Thanks

EQuimper, 2017-10-18:
+1 For this one, get same problem

kosiakMD, 2017-10-18:
@EQuimper, do you use React Navigation (https://reactnavigation.org/) ?

EQuimper, 2017-10-18:
Yes, I figure out a way for me to make sure I dont stack to much the navigation by using a state for the code.

EQuimper, 2017-10-18:
```js
  _onBarCodeRead = (e: Code) => {
    console.log('data code', e)
    if (this.state.vinNumber.length !== 0) {
      return undefined
    }
    if (e && e.type === Camera.constants.BarCodeType.code39) {
      return this.setState({
        vinNumber: e.data
      }, () => {
        if (Platform.OS === 'ios') {
          Vibration.vibrate(500, false)
        } else {
          Vibration.vibrate([0, 500], false)
        }
        this.props.navigation.navigate('MyRoute', { code: this.state.vinNumber })
      })
    }
  }
```

kosiakMD, 2017-10-19:
@EQuimper, I also used similar, but my Barcode Scanner it on single Screen, so I use Component Class internal property, on each `componentWillMount` set as `false` and after first `onBarCodeRead` set as `true`, after read it go back and unmount the Screen

DeividVeloso, 2017-11-01:
Same problem here, in my case I put a if condicion to read only ITF type. But When I move to another screen the Camere keep runing in bacground and spend much battery.

How can I close de Camera if I use Stack Navigator?

kosiakMD, 2017-11-02:
@DeividVeloso Stack Navigator Unmount Screen if you go Back, but if you go forward it is still mounted
1) solution: go back by `backAction`
```
backAction = NavigationActions.back({});
navigation.dispatch(backAction);
```
2) Listen changing of Screens by `onTransitionStart` and `onTransitionEnd`, you will receive Screen (Route) Name (index), and call some Redux Action to unmount Camera

3) just unmount by Redux Action without any additional manipulation

DeividVeloso, 2017-11-03:
@kosiakMD thanks to reply, but I need that screen stay in stack, because my user can back to camera, I have a payment flow, and in the middle I turn on the camera.

Thanks for your help.

ORoberts221, 2018-04-23:
Is there any word on a fix for this? I'm having to look at some pretty ugly work arounds to make sure the callback only fires once, but so that I can still scan another bar code if I navigate back to the camera.

mopx, 2018-06-29:
My solution is to change the barCodeTypes property on the first read.

Change it from the types you're using e.g. [RNCamera.Constants.BarCodeType.qr] to an empty array [].

That will stop the camera from reading the code multiple times. Change them back when you are done.

thakshak, 2018-11-02:
suggestion for react-native-camera to develop a function like
onceBarCodeRead()
to read it only once which is very useful.

marlti7, 2019-05-05:
same problem

blaues0cke, 2019-07-04:
In my opinion this is not a bug since there may be multiple barcodes in the visible area of the camera and react-native-camera cannot (really) determine those positions to remember a already recognized barcode.

MateusAndrade, 2019-07-15:
> In my opinion this is not a bug since there may be multiple barcodes in the visible area of the camera and react-native-camera cannot (really) determine those positions to remember a already recognized barcode.

To solve this i simply "closed" the camera after scanning the first barcode. There is a lot of workaround to handle this. But as you said its a not a bug on react-native-camera, since it only scan the available area, and dont handle cases like that.

closed 2019-07-15 by MateusAndrade

piolAtif, 2020-07-30:
I have recently faced the same issue of multiple scanning by `react-native-camera`  even when navigating to the next screen in the stack navigation. 
I used `event listener` for focus and blur on navigation for solving this issue.
Code:
`const WINDOW = Dimensions.get("window");
class CameraScreen extends Component {
    constructor(props) {
        super(props);
        this.state = {
            isFocused: false
        }
    }

   componentDidMount() {
       this.unsubscribeFocus = this.props.navigation.addListener('focus', () => this.setState({isFocused: true}))
       this.unsubscribeBlur = this.props.navigation.addListener('blur', () => this.setState({isFocused: false}))
   }

   componentWillUnmount() {
       this.unsubscribeFocus();
       this.unsubscribeBlur();
    }

  onBarCodeRead = (event) => {
    const {isFocused} = this.state;
        if(isFocused) {
          ...logic after scanning...
        }
  }

  render() {
  return (
            <SafeAreaView style={styles.container}>
            <RNCamera
                        style={styles.preview}
                        onBarCodeRead={this.onBarCodeRead}
                    />
            </SafeAreaView>)
  }

const styles = StyleSheet.create({
    container: {flex: 1, flexDirection: 'column'},
     preview: {
        width: '100%',
        height: WINDOW.height / 2,
        alignItems: 'center',
        overflow: 'hidden',
        justifyContent: 'center',
    },

})
`