react-native-camera/react-native-camera #951: Camera preview slow to appear, or doesn't appear at all

moksamedia, 2017-10-19:
I'm using React Navigator, and I found this issue: https://github.com/lwansbrough/react-native-camera/issues/518

But even using the setTimeout hack, I still find that the camera preview is slow to appear or doesn't appear at all.

Is this a known problem? Any suggestions?

Node 8.7.0

"dependencies": {
    "apisauce": "^0.14.0",
    "async": "^2.5.0",
    "color": "^2.0.0",
    "format-json": "^1.0.3",
    "jwt-decode": "^2.2.0",
    "lodash": "^4.17.2",
    "moment": "^2.19.1",
    "numeral": "^2.0.6",
    "prop-types": "^15.5.10",
    "querystringify": "0.0.4",
    "ramda": "^0.24.1",
    "react": "16.0.0-alpha.12",
    "react-native": "0.47.2",
    "react-native-animatable": "^1.2.4",
    "react-native-auth0": "^1.2.0",
    "react-native-camera": "https://github.com/lwansbrough/react-native-camera.git",
    "react-native-config": "^0.6.0",
    "react-native-drawer": "^2.3.0",
    "react-native-elements": "^0.17.0",
    "react-native-masked-text": "^1.6.2",
    "react-native-onesignal": "^3.0.6",
    "react-native-permissions": "^1.0.1",
    "react-native-twilio-video-webrtc": "https://github.com/blackuy/react-native-twilio-video-webrtc",
    "react-native-uuid": "^1.4.9",
    "react-native-vector-icons": "^4.3.0",
    "react-navigation": "^1.0.0-beta.7",
    "react-redux": "^5.0.2",
    "redux": "^3.6.0",
    "redux-persist": "^4.1.0",
    "redux-saga": "^0.15.6",
    "reduxsauce": "0.4.1",
    "seamless-immutable": "^7.0.1"
  },

goel135, 2017-10-25:
Any update on this ??

iworkinprogress, 2017-12-18:
Running into this same issue. Seems to be random wether the camera will appear.

moksamedia, 2017-12-18:
I believe this is related to the component not unmounting and remounting when using navigator. I've been using a reset action (in navigator) to return to the page, which forces a dismount and seems to be working.

sibelius, 2017-12-18:
you can use 

```jsx
InteractionManager.runAfterInteractions(() => {
})
```

closed 2019-06-08 by stale[bot]